bitdump/BLHeli #35: Motors accidentally start spinning on cut PWM impulses

Clouded, 2015-01-12:
Hi everyone!

I'm using Emax 20A modules with BLHeli firmware and noticed a strange motor behavior. 
Motors start spinning on short PWM impulses:
 < 1000uS - off
< 780uS - slowly spinning and turning off
< 480uS - spinning wildly

If my guess is correct the ESC thinks that these PWM impulses come in high-frequency PWM mode (which they are not, the frequency is still 400Hz). 

That seems a little bit dangerous because if you unplug the PWM signal cable during the PWM generation the impulse gets cut and the motor starts spinning. I've just observed such behavior which made me investigate that issue. 

1. I'm sending 1000uS impulses, the motor stays still.
2. I unplug the PWM cable, the impulse gets cut, motor starts spinning for half a second.
3. I plug the cable, it also results in cut impulse and the motor spins again for half a second.

[image]

sskaug, 2015-01-12:
Is this with the FW that the ESCs came with? Which I believe is a derivative of 10.2.
There have been some code changes since this to address such issues.

Clouded, 2015-01-13:
Yes, the firmware is the one it came with.
I'm not able to update the firmware and try the new version at the moment as I don't posses toolstick or arduino, but I'll try some time soon. 
I'll close the issue for now.

closed 2015-01-13 by Clouded