bitdump/BLHeli #36: Can BLHeli Be Used to Control Gimbal Motor

arlarson-tuska, 2015-01-20:
Hi I'm trying to build a brushless gimbal and I'm trying to come up with a way to control a Quanum 2208 gimbal motor bidirectionally. I've managed to flash a turnigy plush 6a with the multirotor version of BLHeli and I've set the motor direction to bidirectional. I am able to control it using a pwm signal but when the motor is stopped there is a 1-2 second spool up time where the motor slowly accelerates and has almost no torque, after which it runs fine until it comes to a stop. I've played with the various settings but can't seem to improve it much. I'm hoping to be able to quickly switch the speed and direction of the gimbal motor. Is this even possible with BLHeli? Thanks!

sskaug, 2015-01-21:
Now, I don't know much about gimbal motors, but I thought maybe they are normally operated as stepper motors? And I'm pretty sure brushless motor ESCs are not suited to drive them.

arlarson-tuska, 2015-01-21:
As far as I know the motor I'm using is a 14 pole 114kv brushless motor. You would typically buy a gimbal control board which drives the motors using an L6234d 3 phase brushless motor driver IC. The reason I'm not trying to avoid buying a completed board is because this is an educational research project for my university. The last step of the project is finding a reliable way control the motors. Not sure if blheli will work or not for this but either way thanks for the help sskaug!

zobotek, 2015-01-21:
Try to start with open source gimbal SW  instead of propulsion controller SW.   It's available.   The SW goals are precision positioning and holding torque so the motors are higher Z and the modulation algorithm is quite different. I know of no reason why standard BMC HW would not work.   Typically accelerometers and gyros are added for stabilization unless the platform is rock stable.

Sent via the Samsung GALAXY S®4, an AT&T 4G LTE smartphone

sim-, 2015-01-21:
Gimbal motors are just brushless motors with many turns to give them high enough inductance that a reasonable PWM range from a typical battery voltage can be used, and are driven sinusoidally and blindly (without any motor feedback). There are some cases of ESCs being used to drive them, including some firmware for, e.g., bs_nfet.hex-compatible ESCs (from OlliW, see http://www.olliw.eu/2013/brushless-gimbal-direct-drive/), but the hardware is kind of not really right for the purpose, since they barely needs to support a few Amperes, and you need to support one motor for every axis, which is probably better done with a single MCU that has a lot of hardware PWM channels or similar.

arlarson-tuska, 2015-01-22:
Zobotek - do you have any links to open source gimbal SW? Right now we are using an arduino pro mini with an mpu6050 for the IMU.

Sim-  At the moment we are using an Arduino pro mini for our IMU. Wouldn't we need some sort of transistor for the signals to the motor. The pro mini as far as I know can only supply 40mA per pin which wouldn't be enough to drive it directly. It looks like OlliW is using a TC4427 which I'll take a look at. 
Thanks for the help guys.

zobotek, 2015-01-23:
I can’t give you better info than you could get from a search engine.

Zobotek

closed 2015-01-30 by sskaug