bitdump/BLHeli #41: Burned motor and ESC on Damped light

pinoycoders, 2015-03-12:
I was testing the damped light mode on my Sunrise BLHeli 30a esc since the manual says it supports damped light mode but when I connected my motor and repeatedly gave it some throttle bursts then the motor suddenly smoked (SUnnysky X3108S 720kv with 1355 t-type prop). I quickly remove the battery and unplug the motor I then connected the battery again and the esc quickly let magic smoke out (Motor is not connected). What could have happened here? The motor was not stuttering when I was testing it, throttle response was actually good but then the motor burned. Is it because I was using a 13inch prop?

BLHeli version: 13.1
Battery: 4s
Motor: SunnySky X3108S 720kv
Prop: Tarot T-Series 1355
ESC: Sunrise Slim BLHeli 30a 

[image]

sskaug, 2015-03-12:
This ESC is capable of running damped light. Possibly there is a lot of braking energy from a 13inch prop that could stress the ESC? Or it could be some other random cause for the smoke. Hard to say. Also, I do not know this motor, whether it is challenging with respect to staying in sync or not.

pinoycoders, 2015-03-13:
Is there any other setting that if paired with damped light could possibly burn the motor or esc. Thanks for responding. Anyways I just flashed the rest of my ESC and its now working better than the stock firmware that came with this esc. I would probably stay on the default settings just to be safe :)

closed 2015-04-06 by sskaug