bitdump/BLHeli #44: ESC's not starting correctly each time

marnovdm, 2015-03-19:
Hello,

I flashed BLHeli 13.1 on my Afro 12A Lite v3 ESC's (default settings so far) and using OpenPilot 15.01 I never had any issues. However in OpenPilot 15.02 they added support for OneShot125, and since I upgraded to OpenPilot 15.02 and enabled that option I'm sometimes having trouble getting all motors to start correctly. I posted about this on the OpenPilot forum and someone else confirmed the issue so I thought I'd open an issue for this on GitHub. See OpenPilot thread here: https://forums.openpilot.org/topic/55149-oneshot125-support/?p=412578

Short description of setup:
- CC3D running OpenPilot 15.02
- Afro 12A ESC's, BLHeli 13.1 default settings. Ground and signal connected on all ESC's. One connected with 5v to power FC/RX. Damping not enabled yet (unsure if it works with my motors..)
- Multistar Elite 2300kv motors

Issue description:
When I apply power to my quad, the quad plays it's initial startup beeps (a remote controlled beeper + battery beeper + ESC's etc). Sometimes (I'd say 2 out of 10 attempts) when this is done one or more ESC's make an additional beeping sequence. When I arm and then apply some throttle, one or more motor's then won't start at all. I've already tried recalibrating the ESC's and raising the minimum starting power in GCS. Sometimes when I disarm and wait for a while, the ESC's beep another time and usually I can then start additional or all motors. When all motors start correctly there is no issue flying and OneShot125 seems to work.

Steps to reproduce:
I've been unable to reliably reproduce this, it's always different combination of motors that show an issue. Ea. sometimes front left, sometimes all on the right, sometimes only rear.. It will work correctly 8 out of 10 times. I think this could perhaps be an issue with the protocol detection at ESC startup in combination with heat/timing unreliability? It seems to occur more often after a couple of flights.

sskaug, 2015-03-19:
Hi!

Occasionally I've seen similar reports, and in the code I found a variable that isn't properly defined during input signal qualification. I  have made a fix, that I can send you to try if you want?
Any email I can send to?

Thanks,
Steffen

marnovdm, 2015-03-19:
Hi Steffen,

Thanks for the quick response! This is my first DIY quad so I don't have lot's of extra parts or a spare quad right now and would be grounded if I fried all motors/ESC's. Is there a risk of irreversible damage in trying that firmware (I don't have a USBASP programmer right now, only Afro USB Linker from HobbyKing)? Otherwise I'd be happy to try, though it could be a day or 2 before the weather clears up over here..

Greets!

sskaug, 2015-03-19:
There should be no more risk in updating than it is with a normal flashing or parameter change. The AFRO USB linker should work fine. And probably you can see the effect indoor (just when arming), even if the weather is crappy :).

marnovdm, 2015-03-19:
Okay that sounds great! Could you send it to marno.vandermolen at gmail dot com? I only updated through BLHeliSuite so far, any difference applying this custom firmware file?

And while I have you here, if you don't mind: would you think I'd have any issue enabling Damping with these Multistar Elite 2300kv's? The Afro 12A's should support it but I'm still a bit worried about blowing them up.

marnovdm, 2015-03-19:
For the record I want to note that I've tested the firmware sskaug kindly provided to me and it seems to greatly improve the situation, though some ESCs still take a bit longer to start (I'm guessing some signal detection/timing issue).

marnovdm, 2015-04-02:
I'm currently testing a 2nd test firmware (Steffen calls it 'No25to12') which so far seems to resolve the issue! I'll do some more testing but they each initialize at exactly the same time now.

jkoljo, 2015-04-03:
Great to hear. I am experiencing the same initialization problems you described

sskaug, 2015-04-06:
Thanks for all the input and testing! I will fix the uninitialized variable and change the number of consecutive accepted signal periods from 25 to 12 in the next rev.

jnrabreu, 2015-04-07:
Hi!
I have the same problem. When will the new release with fix be available ?

Thanks

marnovdm, 2015-04-08:
I've done a lot more flying since my last update here and it's still 100% stable like this. Not sure when it will be released.

sskaug, 2015-04-08:
Well, that fix is easy, and ready for release. But there are other things that will be included in the release as well. Try to make it this week, otherwise it will be a couple of weeks.

sskaug, 2015-04-11:
Rev13.2 is now released, with this fixed.

marnovdm, 2015-04-12:
Great news! I'm sure a lot of Afro owners will be happy :D

Changelog:
- Still tweaking startup to make it more reliable and faster for all ESC/motor combos
- Increased deadband for bidirectional operation           - Relaxed signal detection criteria
- Miscellaneous other changes

Would it be worth it for me to upgrade from the no25to12? Not sure what these misc changes are. Thanks!

sskaug, 2015-04-13:
If your setup woeks without problems, you're fine.
There are no general performance improvements, just changes to make it more robust.

marnovdm, 2015-04-13:
I'll keep using the No25to12 firmware then, it's still performing perfectly stable. Thanks!

closed 2015-04-13 by marnovdm