bitdump/BLHeli #45: BL heli 13.1 Multi reversible mode (neutral position = PPM center throttle)

StuttgartRalf, 2015-03-23:
Because the ESCs with Silabs processor do mainly use the internal resonator with +-2% tolerance, the adjusted PPM center throttle position (motor off or neutral position) is not very accurate and it also moves, because of temperature drift. So after a flight it can happen, that a motor doesn't stop running any more and it's still runs slowly.
The easeast way to solve this problem would be to increase the dead zone (use a wider neutral zone) to the adjusted neutral position +-20µs or if you don't like a fixed wider neutral position the width of the neutral position could be also adjustable. At least +-10µs should be applied.
Could you do this with the next update, please?

sskaug, 2015-03-23:
I understand that this can be an issue, a deadband is needed, but it should be as small as possible. Currently it is +-5 throttle steps, which corresponds to approximately +-20us for a 1-2ms input signal. Theoretically, this should allow for the drift in the MCU clock. Are  you sure your neutral is centered in the deadband?

StuttgartRalf, 2015-03-23:
Dear Steffen,

thank you very much for your fast answer.
The problem is, if you use the reversible mode, you can not learn the 
transmitter stick travel from the transmitter any more and so I have to 
adjust it in the menu.
But this adjustment is not accurate enough. The tolerance of +-2% of the 
internal resonator is too much.

I adjusted 1500µs for the center, but the ESCs do need 1484...1496µs and 
this is alread too much tolerance. When the ESCs get warm, it's getting 
worse.

If you have +-800 throttle steps, 5 steps would be only 2.5µs.

+-2% is already 16 steps and we need some reserve, too.

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver. 

position = PPM center throttle) (#45)

sskaug, 2015-03-23:
This is subtracted at a point where the throttle signal has +-255 steps. So +-5 steps is about +-2%. When I measure one ESC, I get 1480-1496us as a stopped range, which is about +-8us out of +-400us and matches pretty well.
Now, with different ESCs being calibrated differently, I understand it can be difficult to get all motors in a copter to stop.
Increasing this value in the next rev is no problem, it's just  a number.
But what are the tradeoffs in increasing it? How high do we want to go?
I don't think this is worth an additional adjustment parameter.

StuttgartRalf, 2015-03-23:
Dear Steffen,I think we should try the double value +-10steps = +-16us.Mit freundlichen Grüßen / Best regardsRalf HelbingGeschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische InformatikGraupner/SJ GmbHHenriettenstr. 96, 73230 Kirchheim / Teck, DeutschlandTel: +49 (0)7021 722-185Mobil: +49 (0)173/5985202Fax: +49 (0)7021 722-199Mail: r.helbing@graupner.deWeb: http://www.graupner.deGraupner/SJ GmbHSitz der Gesellschaft: Kirchheim unter TeckAmtsgericht Stuttgart HRB 744445Geschäftsführer / Managing Director: Ralf HelbingDiese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder rechtlich geschützte Informationen. Sie ist ausschließlich für den Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit Viren befallene E-Mails entstehen.This e-mail and any attachments contains confidential and / or legally protected information, and is intended exclusively for the person to whom it is addressed. If you are not the intended recipient, or if this e-mail has been sent to you mistakenly, please inform the sender immediately, and destroy this message. It is prohibited for any unauthorised person to read, copy or store this e-mail in any form, to pass it on to a third party, or to handle or process it in any other way. We use current virus protection programs. Nevertheless, we deny all liability for damage which may result from e-mails containing viruses which are sent to the receiver. 
sskaug --- Re: [BLHeli] BL heli 13.1 Multi reversible mode (neutral position = PPM center throttle) (#45) --- Von:"sskaug" <notifications@github.com>An:"bitdump/BLHeli" <BLHeli@noreply.github.com>Kopie:"StuttgartRalf" <r.helbing@graupner.de>Datum:Mo., 23.03.2015 22:22Betreff:Re: [BLHeli] BL heli 13.1 Multi reversible mode (neutral position = PPM center throttle) (#45)This is subtracted at a point where the throttle signal has +-255 steps. So +-5 steps is about +-2%. When I measure one ESC, I get 1480-1496us as a stopped range, which is about +-8us out of +-400us and matches pretty well.
Now, with different ESCs being calibrated differently, I understand it can be difficult to get all motors in a copter to stop.
Increasing this value in the next rev is no problem, it's just  a number.
But what are the tradeoffs in increasing it? How high do we want to go?
I don't think this is worth an additional adjustment parameter.

sskaug, 2015-03-23:
So be it then :).
+-10 steps on the next rev.

Cheers,
-S

StuttgartRalf, 2015-03-24:
Dear Steffen,

thanks a lot. I would be happy, if you could inform me after the release.

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver.

StuttgartRalf, 2015-04-09:
Dear Steffen,

I hope you had a happy easter holiday.

When do you plan to bring the 13.2 update with the wider deadband?

We like to test it soon for our new coming product, if possible.

Thanks a lot.

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver.

sskaug, 2015-04-11:
Implemented in rev 13.2, which is now released.

StuttgartRalf, 2015-04-11:
Dear Steffen,

thanks a lot.

Will the v13.2 update be available for the EMAX 20 A, too?

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver.

sskaug, 2015-04-11:
All supported ESC files are updated.

StuttgartRalf, 2015-04-12:
Dear Steffen,

is the bootloader automatically installed with 13.2?

How can I get it work? Which device do I need to use the bootloader?

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver.

sskaug, 2015-04-13:
With the SiLabs code, the bootloader is always included.

Basically the same sticks work, that work with the Atmel BLHeli bootloader.
But, unfortunately, many of the SiLabs ESCs have a high input resistance. So, sticks with a low pullup resistance will not work.
Currently an FTDI with a high pullup will work, and an Arduino.
Achim will come up with a new version of the Suite shortly.

StuttgartRalf, 2015-04-13:
Dear Steffen,

thanks a lot, great.

I found a problem on the 13.2 for the EMAX 20 A.

With 13.1 the EMAX 20 A worked fine with High Motor Timing and 2204 2300KV 
motors.

With 13.2 the EMAX 20 A does have problems at medium to high rpms. The rpm 
is not consistant any more and timing errors with ESC cut-offs are 
possible.
With Medium High Timing it seem to work same or similar as with 13.1 with 
High timing.

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver.

StuttgartRalf, 2015-04-13:
Dear Steffen,

P. S. 

the Atmel ESCs seem not have the problem at High timing.

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver.

StuttgartRalf, 2015-04-13:
Dear Steffen,

the EMAX 12 A does have the same problem. I have to use Medium timing, 
even medium high is too much.

Maybe all Silabs could have this problem now.

All Atmels seem to work better now.

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver.

sskaug, 2015-04-14:
I'm surprised you see much difference for SiLabs, there were only small changes to the code.
What make of motor do you use? What battery? Prop?

StuttgartRalf, 2015-04-14:
Dear Steffen,

I'm surprised you see much difference for SiLabs, there were only small 
changes to the code.

What make of motor do you use?
EMAX 2204
Graupner 2209

What battery? 3S 2200 mAh

Prop? Graupner 3D Prop 6 inch
 

Could you explain what exactly was changed? Or could you change it back 
for the Silabs?
The ESCs have now timing errors.

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver.

StuttgartRalf, 2015-04-14:
Dear Steffen,

the Hobbywing XROTOR with Silabs seem not to have the problem like the 
EMAX with 13.2 firmware.

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver.

sskaug, 2015-04-14:
Thanks for the info. From what I've seen, the EMAX motor is benign, and 3S is not pushing it. So this should work very well. Before I released the code, I tested a Skywalker 20A on 4S with various motors and all was well.
I'll look more into this when I get home this weekend, I'm currently travelling.

StuttgartRalf, 2015-04-16:
Dear Steffen,

Afros and a few other ESCs use the ICP1 pin for signal input. This enables 
use of HW for measuring input pulse length, and therefore they run more
smooth with OneShot125 input signal.

Could you please tell me, which other ESCs do also use the ICP1 pin for 
the signal input.
What about the ZTW and SUNRISE?

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver.

sskaug, 2015-04-16:
It is in the "supported escs" doc. As far as I recall, the Afros, an RCtimer and the dys sn20a

-S

StuttgartRalf, 2015-04-18:
Dear Steffen,

does the oneshot mode works also with the reversible mode?

If yes, how about the neutral zone. Also +-5% from the middle?

Mit freundlichen Grüßen / Best regards

Ralf Helbing

sskaug, 2015-04-19:
Hi Ralf

Thanks for the find!
I have now looked into it, and it appears to be a weakness in interrupt enabling and disabling. That is there for all 13.x SiLabs codes, and becomes a visible at high timing. The differences you have seen between 13.1 and 13.2 may be more subtle and due to other minor changes.
This issue does not exist in Atmel codes.

Would you help verify that this actually fixes what you have seen? You have an email I can send test code to?

Also, the bidirectional deadband shall be the same percentage for OneShot as for normal input signal.

Thanks,
Steffen

sskaug, 2015-04-19:
...just saw your email already listed above....
I'll send you some code.

MohamedElwazzan, 2015-06-07:
Hi Steffen,

Could you Please compile revision 13.2 for the vendor specific versions/EMAX 40A, because I have a problem with deadband for bidirectional operation

thanks
Elwazzan

sskaug, 2015-06-07:
Hex for the EMAX 40A should be in the 13.2 release. But there will be significant mprovements to bidirectional in the next release. Which is pri one for me at the moment.

MohamedElwazzan, 2015-06-07:
Thanks for your reply,
Now I use the hex file in that link https://github.com/bitdump/BLHeli/tree/master/Vendor%20specific%20versions/EMAX

this works fine but it has a very small deadband for bidirectional

I know that there is 13.2 release in that link:
https://github.com/bitdump/BLHeli/tree/master/SiLabs/Hex%20files

and it has a wider deadband
but it has a bug when I start the motor after a stop condition the ESC resets it self I think it makes a short circuit for a very small time !!!
After reset it works fine until I send a stop condition and start again.

so I use EMAX_BLHeli_40A_V13.1.HEX
https://github.com/bitdump/BLHeli/blob/master/Vendor%20specific%20versions/EMAX/EMAX_BLHeli_40A_V13.1.HEX

Is there anyway to modify the deadband in that version ?

Thanks

sskaug, 2015-07-08:
Now, I've spent considerable effort on bidirectional and startup lately.
And now rev14.0 is posted, with what I think are significant improvements.
Still, bidirectional operation is largely dependent upon ESC HW quality.
If there is comparator offset or bemf sensing network mismatches, this will degrade startup reliability significantly. Closing this issue.

closed 2015-07-08 by sskaug

StuttgartRalf, 2015-07-09:
Dear Steffen,

do you know when the BL Heli Suite 14 will be available?

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver.

4712, 2015-07-09:
Vermutlich Freitag. Bitte noch ein klein wenig Geduld...

Achim

4712, 2015-07-12:
BLHeliSuite 14.0.0.0 is released.

StuttgartRalf, 2015-07-13:
Dear Steffen,

there seem to be a problem with the Afro 20A Multi.hex Version 14.0

The motors do not start.

If I go back to the 14 beta version, it works correctly.

Mit freundlichen Grüßen / Best regards

Ralf Helbing
Geschäftsführer / Managing Director Dipl.-Ing. (FH) Elektronik/Technische 
Informatik

Graupner/SJ GmbH
Henriettenstr. 96, 73230 Kirchheim / Teck, Deutschland

Tel:     +49 (0)7021 722-185
Mobil:     +49 (0)173/5985202
Fax:    +49 (0)7021 722-199
Mail:    r.helbing@graupner.de
Web:   http://www.graupner.de

Graupner/SJ GmbH
Sitz der Gesellschaft: Kirchheim unter Teck
Amtsgericht Stuttgart HRB 744445

Geschäftsführer / Managing Director: Ralf Helbing

Diese E-Mail sowie eventuelle Anhänge enthalten vertrauliche oder 
rechtlich geschützte Informationen. Sie ist ausschließlich für den 
Adressaten bestimmt. Falls Sie nicht der beabsichtigte Adressat sind oder 
falls diese E-Mail irrtümlich an Sie gelangt ist, informieren Sie bitte 
sofort den Absender und vernichten Sie diese E-Mail. Das unerlaubte Lesen, 
jegliche Vervielfältigung, Speicherung, die unbefugte Weitergabe und 
sonstige in diesem Zusammenhang stehende Handlungen sind untersagt. Wir 
verwenden aktuelle Virenschutzprogramme. Gleichwohl schließen wir jede 
Haftung für Schäden aus, die dem Empfänger durch von uns zugesandte mit 
Viren befallene E-Mails entstehen.

This e-mail and any attachments contains confidential and / or legally 
protected information, and is intended exclusively for the person to whom 
it is addressed. If you are not the intended recipient, or if this e-mail 
has been sent to you mistakenly, please inform the sender immediately, and 
destroy this message. It is prohibited for any unauthorised person to 
read, copy or store this e-mail in any form, to pass it on to a third 
party, or to handle or process it in any other way. We use current virus 
protection programs. Nevertheless, we deny all liability for damage which 
may result from e-mails containing viruses which are sent to the receiver.

sskaug, 2015-07-13:
I'll take a look at it. For this rev I had to do manual edits in all the
ESC specific inc files...

4712, 2015-07-13:
confirm: seems the common for all Afro