bitdump/BLHeli #47: Compiling BLHeli on GNU/Linux

mashu, 2015-04-06:
Hi,

How to compile BLHeli on GNU/Linux, because BLHeliSuit is for windows only ?

Thanks,

marnovdm, 2015-04-06:
It seems like you're confusing 2 things; compiling the firmware from source and using BLHeliSuite to flash it to your ESCs. I don't know how to flash BLHeli from Linux either (I dualboot to Windows for it at the moment, though I'd really like a native BLHeliSuite for Linux as well). I don't think you need  to manually compile the firmware unless you modify things in the source obviously.

mashu, 2015-04-06:
As far as I can see BLHeliSuit can configure firmware and reconfiguring usually requires recompiling, unless they have binaries for every setting, but making that would be time consuming. I am aware that BLHeliSuit is only for windows, hence I asked about possibilities of compiling BLHeli (not suit) on GNU/Linux ;)

marnovdm, 2015-04-06:
Alright sorry I misunderstood your question :-) With BLHeliSuite on Windows you flash the firmware once, and then you can change settings through the tool, which is way faster than flashing. I don't think it's changing hardcoded values in source and recompiling & then uploading that firmware, I think it has a settings register of some sort which it just updates.

mashu, 2015-04-06:
I will stick to what I have now, but it should be possible to quickly sketch gtk ui in glade, make callbacks and simple app in python. Too bad firmware cannot be compiled without involving commercial software, I usually lose interet in such cases.

bitdump, 2015-04-06:
Well, it is only 8051 asm you are talking about right? Linux has pretty good support for AVR chips. I assume it is not a problem for AVR ESC. I have look into this problem. I think there is possibility to compile the 8051 asm in Linux. All we need is a pre-processor convert the Ride IDE asm into SDCC 8051 asm. I think it can be done in python or some thing like that.

bitdump, 2015-04-06:
Regarding flashing BLHeli in Linux. It can be done with some hacking. Linux kernel already have some support for the C2 protocol. See http://lwn.net/Articles/304330/
It just need some hacking if you have a Linux can access to the GPIO pins, e.g. RPi.

mashu, 2015-04-06:
I am sorry, I tried compiling Atmel code, but I think I know (stupid me) where the error comes from :)
I am using "avra" but I didnt declare device type. Should be possible then to flash it via servo with RapidFlash plugin for GoogleChrome I guess? My ESC have bootloader.

mashu, 2015-04-21:
Ok it was possible to flash BLHeli using BLHeliSuit running in wine after linking
ln -s /dev/ttyUSB0 ~/.wine/dosdevices/com1
wine BLHeliSuite.exe

closed 2015-04-21 by mashu

WZ9V, 2015-08-28:
mashu what adapter are you using?

I tried your suggestion with an Arduino Uno (ttyACM0) and I still could not get it going.

mashu, 2015-08-29:
I used AfroUSB Programmer. Afro ESC mcu are Amtel AVR chips you should be able to program them with arduino as well but I didnt try that route myself.

0xCCD, 2015-10-15:
@mashu have you installed any drivers for AfroUSB under Windows ? 
i'll try the same but with an DYS USB-Linker i already tried the symlink, but unfortunately it don't get a connection !