bitdump/BLHeli #55: OneShot signal with update rates of 1 kHz and higher

Trex4Git, 2015-04-21:
As i have already written in the rcgroups thread i want to test OneShot input signal with higher update rates.
The problem is that input signals with a frequency of 1kHz or above gets detected as 0-100% PWM.

To use a high control rate in the FlightController and also a high frequency of the ESC signal, i tried to activate the OneShot mode (which syncs the ESC output to the control loop) but instead of using the 125-250µS Oneshout pulses, i used 0-80% PWM signals corresponding to the control loop rate (together with a Motor Gain of 1.25) .

For the 1kHz setup it seems to work. The ESC arms with this signal and reacts to the throttle.
For the 2kHz setup it doesn't arm. When armed with a normal 2kHz PWM, and after that switched to OneShot mode the ESC reacts to the throttle.

The arming problem maybe is caused by a jitter in the frequency of the OneShot signal. 
As some FC software like Openpilot or Taulabs don't use an internal timer for the timing of the control loop but triggers them by an gyro interrupt. 

I am using Taulabs and looked at the ESC signal with a logic analyzer.
Most times the frequency is constant, but sometimes for 1 or 2 pulses there is a change in the frequency which is sometimes so large, that BLHeli might detect it as a different PWM frequency.
This might be the problem with the arming. 

And when BLHeli detects the frequency not only at the arming for the correct 0-100% scaling, but also during normal work, additional problems could also happen during normal operation/flight, as the pulse width stays the same, only the frequency changes for 1 or 2 pulses.

Are there features planed for the next versions of BLHeli to support OneShot with higher update rates?
A selection between auto detection of the signal type, or one of the fixed input types (PPM, OneShot125, 1kHz PWM, 2kHz PWM, 4kHz PWM, 8kHz PWM, and 12kHz PWM) would offer high degree of freedom for advanced users.

Additionally a higher resolution of the sampled input signal would be nice. I don't know if this could be possible with just using a higher clock on the used timer module, and a rescaling of the values.

zobotek, 2015-04-21:
Is it time to abandon PWM on the flight controller<---->ESC interface completely and go digital (e.g., PCM)?

Regards,

Alex

sskaug, 2015-04-21:
Adding a BLHeliSuite setting that locks the input to PPM/Oneshot125 should be easy (maybe for a rev 14?). If there is available code space, that is.
Instead of fiddling like you do, I'd be happy to send you a test code. Just let me know which ESC and your email.

Trex4Git, 2015-04-21:
As this will be my first quad, i have not so much  experience yet, but want to try new stuff. I am a electrical engineer so some fiddling is ok ;), and these test didn't consume much time. And not all components for the quad have arrived yet, so some bench testing and getting more information at the moment.

But i would also be happy to test your code. At the moment i am using the H-King 10A ESC.
(But looking forward to new ESCs. If the Bearhug 18A ESC would be more reliable they seem to be good candidates to me.)

My mail is" Trex at gmx-topmail.de ".

Trex4Git, 2015-04-23:
After looking a little deeper into the sources i have noticed that the PWM Output Signal seems to be controlled by a 8bit variable (0-255). So when using OneShot125 with 250 steps, the resolution loss compared to normal PPM Input is not as big as i have thought earlier. And to get a higher resolution just higher sampling rate of the input signal would not be enough, and it seems that a lot more has to be changed.

But as all parts for my copter arrived i can finally build it, and start hopefully flying on the weekend.

And with the autotune function in taulabs  you get also values of the system identification. With the "reaction time" of the system (tau ;) ) i think i could clearly see if higher control rates would bring an advantage.

So getting a test firmware with support for higher update rates with OneShot would be nice.

sskaug, 2015-04-23:
Test code sent

Trex4Git, 2015-04-23:
Thanks,
flashed them and tested without Props.

With 1kHz the ESC Arm and seem to work. 
With 2kHz no arming of the ESC.

sskaug, 2015-04-24:
Ok, I'll have to get me a signal generator...

Trex4Git, 2015-04-24:
I can also do some measuring of the output signal of the FC and post it here this evening.

Trex4Git, 2015-05-05:
Not much free time, and often bad weather, but i did some early tests during the last days. 
At first with the modified 13.2 version for OneShot125, and also some testing with 4kHz 0-100PWM with normal 13.2. 
(With the 0-100% PWM i can test also 666Hz, 1kHz and 2kHz in future with normal BLHeli versions. And  in the next time maybe the PWM output of my FC is enhanced with higher resolution,  than a 8kHz PWM should give the same resolution with the BLHeli ESC as OneShot125, and also should have a short delay)

The tests are my first autotune tests, and i did them only once or twice with the same setup. 
For better reliability of the data i have to do more tests with the same setup in future. 

During autotune a system time constant tau is measured (time between a command from the FC to the ESC, and a measured result in the sensor data):

control loop / gyro rate | signal type | tau
------------- | ------------- | -------------
666Hz | OneShot12 | 59.5ms
1kHz | OneShot12 | 45.9ms
1kHz | 4kHz PWM | 35.3ms

Detailed autotune data is there: http://forum.taulabs.org/viewtopic.php?f=17&t=268&start=80
and more details about the quad: http://forum.taulabs.org/viewtopic.php?f=10&t=682

sskaug, 2015-05-05:
Interesting data. Although surprisingly large changes. About 14ms tau change from 666Hz (1.5ms) to 1kHz (1ms). Is there something else going on?
Also, almost 11ms from OneShot125 to 4kHz PWM is much more than expected.

Trex4Git, 2015-05-05:
Yes i don't know how reliable the data is. I have to do more tests. Maybe 4 or 5 autotune runs with the same setup in the same conditions for better statistics. 

Today i did some more tests with the same setup and i see some bigger differences than expected. 

It was a little bit windy. Had to do some manual control during the autotune, and don't know if that has some influence. Also the battery got empty during the test, and i had to change to another one (after first test for 2kHz from 4s/2200mmAh/25C to 4s/1800mAh/40C).

control loop / gyro rate | signal type | tau
:-------------: | :-------------: | :-------------:
1kHz | 4kHz PWM | 34.5ms, 30.2ms, 31.5ms
2kHz | 4kHz PWM | 34.1ms, 31.2ms, 31ms, 31.4ms

sskaug, 2015-05-06:
I just tested OneShot125 at high rates, and with the next rev of code where PWM input is disabled, it works up to pulse rates at least towards 4kHz. So, this should be possible in the net rev.

Trex4Git, 2015-05-06:
Sounds nice.
But will PWM Input complete removed, or can it be configured in the BLHeliSuite ?

sskaug, 2015-05-07:
PWM input will still be available. I'm thinking of making a "Enable PWM input", with a check box in the Suite. Still pondering on the default, think it will be enabled.

Trex4Git, 2015-05-07:
That is a good idea, so existing user don't have to change anything, to have the same setup as before.

At the moment i think there are not so many users using so high control rates, and when they want to have this advanced feature, the worst thing they would have to do is read the release notes or the manual and rewrite a changed setting to the esc.

Trex4Git, 2015-06-04:
As this seems to be integrated in BLHeli 14 i think this issue can be closed.

But i have read that some people are already testing the new version, and a Screenshot showed already the new BLHeli Suite.

So it would be nice if i could also test this new version. For the high frequency OneShot support, and i noticed some start up problems with BLHeli 13.2.
(H-King 10A ESC, my mail is " Trex at gmx-topmail . de ")

closed 2015-06-04 by Trex4Git