bitdump/BLHeli #62: ESC Signal lost

0n3-70uch, 2015-05-12:
Hello,

I use BLheli 13.2 and have a problem with 3D mode (Quadropter with NanoWii). When changing the direction of rotation one motor fails. It is not always the same motor. The problem does not occur so often with a 3S Lipo battery. But with a 4S battery the problem is very big. 

When I disarm and arm the copter (after a motor fail) the ESC beeps. After that everthing works fine. It looks like a signal failure betwen ESC and flightcontrol.

I have tested the following:
- temperatur protection on/off
- demag compensation off/low
- pwm frequenzie high/low/damped low
- motor timing low/medium/high
- oneshot and normal signal

ESC: 4x Hobbywing 20A
Flightcontrol: Flyduino NanoWii
Motors: T-Motor 2204
Propeller: Graupner 3D 6x3"

Greetings
Fabian

sskaug, 2015-05-12:
So, if I understand you correctly, the failing motor does not start again until after you have disarmed and armed?

0n3-70uch, 2015-05-12:
Sorry for my bad english. I made a short video:
https://youtu.be/KkHTxG6bcrk

You can also hear the motor beep (in the video). Often only the left motor stops, but the others have the same problem.

Greetings
Fabian

sskaug, 2015-05-12:
Thanks for the feedback, I will look into it. Is it XRotor 20A ESCs?

0n3-70uch, 2015-05-12:
Thanks too.

The ESC is called: Favourite 12A 4in1
The hardware is based on Skywalker 20A.

sskaug, 2015-05-12:
Have you googled this ESC? There are plenty of reports that it does not work properly (reboots etc) on 4S....

0n3-70uch, 2015-05-12:
Oh yes. It looks like a hardware problem. I have found another video:
https://youtu.be/QXsZpJCqjrY

0n3-70uch, 2015-05-12:
Here is the solution:

"Hi Everyone, I am Jenny from Favorite Electronics, who is the manufacturer for this ESC. Sorry to cause the reboot issue and late reply. It's hardware problem and the solution is easy, just need to replace a bigger capacitor from current 0.1uF to 1uf. Please refer to my below picture from the link to point us what the cap is. Any questions, please contact with me for jennyxcy2008@gmail.com."
[image]

closed 2015-05-13 by sskaug