bitdump/BLHeli #63: BLHeli for car motors

Luisonson, 2015-05-13:
Hello,
I don't know if this thread can stay here or I should ask in another place. If so, tell me ^_^

First of all, thanks a lot Steffen@sskaug for your MASSIVE work.

I want to use brushless motors for a line follower, but I imagine that my requirements are the same for any kind of precision motor control, not just for my robot.

I want to use the SN20A ESC with the motor http://www.hobbyking.com/hobbyking/store/__9069__Turnigy_1230_Brushless_Inrunner_Motor_4500kv.html
I think this ESC has power enough to deal with this motor.

BLHeli should be able to manage this motor??
Also, no dead-zone should be better. The SN20A has a crystal, so I think no dead-zone will increase the precision. Exist the possibility of  disabling this feature?
Speaking of features... BLHeli has somehow a brake... This brake differs from the esc-car brake? (if I change the signal from 80% to 20%... I want the change as fast as possible).
I was reading this thread (https://github.com/bitdump/BLHeli/issues/55), and be able to update the esc at 2 o 4KHz sounds perfect!! When is predicted to be available to the SN20A ESC??
Which is the precision for a normal signal?? (steps in OneShoot125, for example).
In which circumstances the esc will disarm?? In some scenarios maybe I don't want this feature...

I have several arduinos, so I don't mind test codes ^_^, also, install avrstudio and try to make some changes, it doesn't mind me, but I'm not very good in asm...

sskaug, 2015-05-13:
Not quite sure I understand all your questions. But BLHeli should be able to run the inrunner motor that you linked. As to details on arming and signals, I think the documentation should provide most answers. BLHeli is not very well suited for cars, though, as it does not support hall sensors, and therefore you will not have much torque when starting.

Luisonson, 2015-05-13:
Hello,

Sorry for my English, is not my first language and I explain myself poorly.
Here (https://github.com/bitdump/BLHeli/issues/45), you speaks about the dead-band. I'm asking if the dead-band can be configurable by interface and put it to 0 (no dead-band).
* The esc for cars, usually have several brake intensities, the blheli brake how is regarding those brakes?? Can be configured stronger ones??
* Speaking about... If I set the esc to work in 3D mode, and I send for example a 1750 pulse (forward 50% aprox), and then a 1250 pulse (backward 50%)... how would react the esc?? Reading the code, I think that just try to stop the motor, and then with the motor stopped it puts the signals to move it in the other direction... I'm right?
* When will this feature (https://github.com/bitdump/BLHeli/issues/55) available?
* How many bits did BLheli use for detect a signal? (the resolution/precision of the esc).

I will buy the motor and the esc, also, I have a 1024 encoder, so I will be able to see the signal I send and the exact motor speed ^_^

I will inform you.

Thank you for your time.

sskaug, 2015-05-14:
Deadband can be set to zero by modifying the code.
BLHeli does not have braking as such, but damped light will brake the motor. Currently one braking force, but earlier revisions had a choice of force.
When reversing direction, the motor is first braked (heavily) to a halt, then started in the opposite direction.
Feature #55 will be in the next rev, release date still not known.
Signal detection in BLHeli is generally 500ns resolution.

Luisonson, 2015-05-14:
Thanks a lot for your answers! ^_^

* I can see in the code that RCP_MAX is defined at 255... so, blheli works with 255 steps, BUT, if it's not working in governor mode, blheli multiply the signal by 1.0625, using then just 240 steps... I don't know why :/
* I suppose that the deadband code is just:
	ldi	XL, 10						; Subtract deadband 
It's a petty is not modifiable by software, because are 10 steps less in precision... so finally just 230 steps... and in 3D mode only 115 for each side.

* Seems that blheli captures the pulse with the timer1... this timer is a 16bit-timer, it is impossible to increase the resolution? 

Don't hesitate to point me the obvious, because for me it's really hard to follow asm code.

sskaug, 2015-05-16:
- The x1.0625 is there just to make sure full throttle is always reached on 0-100% pwm input signals, even if there are slight differences in clock speed.
- Yes, that's the deadband.
- If resolution is to be increased, timer clock will have to be increased. The timers are used for various purposes, and differently in Atmel and SiLabs :).

Luisonson, 2015-05-17:
As i bought the sn10a, i'm interesting only in the atmel's side.
When I receive everything (seems in several weeks, even some months :_( ), then I will ask you again :) .

Thanks again for your work ^_^

sskaug, 2015-07-08:
Closing for now

closed 2015-07-08 by sskaug