bitdump/BLHeli #64: E-max 12A - BLHeli: do not start

favoretto85, 2015-05-13:
Hi.

I flashed my 4 emax 12A with blheli and set it up correctly, but it do not work. 

The flashing process was ok. As well, the setting process. 

When a put Naze32 (cleanflight) at normal mode, the esc star normal. But, when Naze32 set up oneshot125, the esc start singing (2 beeps per second continuously) and nothing more happens. I've tried to calibrate the esc with GCS, but ESC do not recognize calibration setup (if it is oneshot125 enable; if oneshot125 is disable, everything works fine, included calibration setup).

So, i would like to know if could some one help me.

Tanks

sskaug, 2015-05-13:
I'll do some more testing with signals from Naze32.
Did you try rev13.2? Did you also try other revs?

favoretto85, 2015-05-13:
I'm using the last cleanflight firmware available to naze32 and cc3d, as well. The problem seems to be at emax/12a. 

The "read" was OK, so I thing that flash is OK, too.

But it still do not work (just with oneshot125 enable). If it is using normal protocols, everything works fine.

There is something usuals that should I see or that cause this problem (jus beeping 2 times/second without arming)?

Tanks

sskaug, 2015-05-16:
There have been some reports on problems with arming. I have tried to reproduce here, but so far not with success. I will try more
But for the next rev, some changes to the input signal detection will be done to make this more robust.

favoretto85, 2015-05-17:
I appreciate your attention. Tanks for everything and good luck.

Marcelo.

sskaug, 2015-05-19:
I just tried an EMAX 12A on Cleanflight, and it operates as it should.
Can you please verify using BLHeliSuite, what code you have on your ESC?
[image]

andreiva, 2015-05-23:
Hi, I also flashed some EMAX 12A's , Turnigy Plush 18A and Afro 30A with BLheli 13.2. After flashing I tried to calibrate them but all i heard was power up tone (three beeps) as esc didn't notice zero thr. All above mentioned esc's had same problem. I initially did this with V8R4-II Frsky receiver, then I switched to V8R7-II (same problem) then I switched to D4R-II (flashed with Firmware-D4R-II_27mS http://www.frsky-rc.com/download/view.php?sort=&down=4&file=Firmware-D4R-II_27mS) and this time I was able to calibrate esc's normally.

favoretto85, 2015-05-23:
Hi.

I'm using Aurora 9. I tried with Optima 7 (pwm) and Minima (ppm). But
nothing happens. All escs start beeping two time per second with some
interval. All of them start beeping at same time, but the timing beep is
different. So, after some cycles everything becomes a big noises, with
everyone beeping at different time.

I do know what to do...

I will make more tests this wknd. After that, I'll tell you what happens.

Tanka for your attention.

Marcelo Oliveira Favoretto
OAB/GO n. 31.719
(62) 8564-8698

favoretto85, 2015-05-25:
Hi. I’ve tried everything that i can thought (i reduced esc’s wire and wire connection between CC3D and RX and tried to do esc calibration with GCS), but nothing happens. ESC still keep beeping 2x/second and nothing more happens… I dont know what to do anymore…

I’ll buy other ones to try…

Tanks.

Marcelo Oliveira Favoretto
OAB/GO nº 31.719
(62) 8564-8698

andreiva, 2015-05-25:
Favoretto85 this is a software problem, which will probably be fixed soon.
No need to buy new escs. I know that Version 12.1 works fine, you can try
it, while waiting for a fix.

favoretto85, 2015-05-25:
Ok. I’ll try it. Tanks for your attention.

Marcelo Oliveira Favoretto
OAB/GO nº 31.719
(62) 8564-8698

sskaug, 2015-07-08:
This should now be fixed in rev14. If not, then please reopen this issue.

closed 2015-07-08 by sskaug