bitdump/BLHeli #67: EMAX Simon Series 40amp fried a startup.

agressiva, 2015-05-15:
Hello, i received 4 emax 40amp simon series ESC.
I surprised when i open the esc do reflash end encountered an silabs mcu.  
The motors i used is the old multistar 4225-610Kv / 16 poles

On the first one i made some test with blheli 13.2 using an 3 battery and every thing work fine.
Very smooth, no sync loss .

When  connect to 6s battery it started and worked fine on the first two tests.
When i put throttle for the third time the esc fried.

I reflash the second esc and this time i used a 50w carc bulb in series with esc.

It start on the first and second attemp. On the third the esc reset. It occur because the esc consume too much current.

Soh i made some more tests.

I see that when i remove te throttle the motor spin down and after some threshold rpm the esc apply some break.

When i try to restart the esc reset.

If i reduce the speed very slow the motor stop and start without problem.

Wen i reduce the speed very fast the esc apply the break and when i restart the esc reset.

I not sure you understand my poor english.
If you like i can make a video.

BTW, i do the smae test with other ESC and the same MOTOR and it worked fine.
can you help with this issue ?

agressiva, 2015-05-16:
the most strange part, after i changed the parts fried on two esc (4 fets, ir2003 and mcu) the problem gone.

sskaug, 2015-05-16:
So, is there any takeaway for users in general?

agressiva, 2015-05-19:
Hi, sorry for mine mistake. The problem is the ESC hardware.

The fet driver of rail A has problem.
I changed the ir2003 of all 4 escs and all work perfect.

agressiva, 2015-05-19:
take care with EMAX 40amp Simon series ESC.

All escs i received come with fet driver of rail A with problem. (ordered on 27/5/2015)
After changed the fet driver all escs work perfect.

closed 2015-05-19 by agressiva