bitdump/BLHeli #68: Emax 12A bad start motor

error414, 2015-05-17:
Hi,
I try blheli on emax 12A simon series and i have problem with start motor (BE1806-2300kv). Before start motor, motor twitches. Please look at this video https://youtu.be/-WeUbMGoha4 ESC 2 is powered by simonk FW, and esc 4 is powered by lattest (13.2) BLheli firmware.    

I tried change timing from low to hard. When i set hard timing, it is better but not good. I tried change PWM frequency/dumped, demag compensation, closed loop mode, motor gain, startup power,  but no efect. 

This is my last settings http://postimg.org/image/l3xz637yj/ same setting have ESC on video.

Configuration:
ESC: emax simon series 12A
Motors: DYS-BE1806-2300KV-Brushless-Motor
Controller: naze32 (openshot125 OFF)
Battery: 3S 

Thank you very much
Petr Cada

sskaug, 2015-05-17:
I've seen the same on some of my EMAX ESCs. I believe there are asymmetries between the three phases, that causes offset in the bemf detection. Probably and possibly, there are mismatches in the resistor dividers for the bemf signals. Probably your ESC 4 has less mismatch.

And the original FW on these is not SimonK, it is BLHeli. An earlier revision (10.2 or 13.1, maybe) that is modified by EMAX, primarily to use their programming card. But afaik they have not changed the startup part of the FW.

The motor does twitch, but after all it starts. That was one of the modifications I did in 13.2, in order to be able to live with these levels of mismatch.

error414, 2015-05-17:
thank you very much for quick response. I will try flash older blheli FW for test. Emax probably will not be good ESC ;).

sskaug, 2015-07-08:
There should nwo be significant (hopefully) improvements on this in rev14.

closed 2015-07-08 by sskaug

peisel, 2015-07-17:
Issue still exist in v14 ... use 13.1 medium high Timing at the moment

Shkolik, 2019-03-05:
This issue still exists :( Emax RS1106 6000kv motor.