bitdump/BLHeli #70: HK UBEC 10A w/ NAZA-M V2 do not initialize

MarenB, 2015-05-19:
Hi there,
I have trouble with my Hobbyking UBEC 10A ESCs connected to a NAZA-M V2:

Firmware 13.2 and PPM min/max are set to 1000µs/2000µs. ESCs do not initialize, I suppose that is caused by the NAZA sending 940µs PPM signal only, when not armed.
To intitialize the ESCs, I have to unplug them from the FC shortly and re-connect them after they make the beep-sound. Afterwards I can start the motors.

sskaug, 2015-05-26:
It might also be related to signal qualification by the ESC. I'm doing changes to that now, if you want test code, email me at sskaug69 at gmail dot com.

reModding, 2015-05-27:
Have exactly the same problem with Naze32 (latest stable Cleanflight) and Blueseries 12A (HobbyKing).

MarenB, 2015-05-28:
Can you confirm that your ESCs do not beep at all after powering them and before you unplug them from the FC?

It seems, at least in my case, they stay in bootloader mode waiting for serial communication.

Test version with modified BL has been sent to me this night, will test it soon.

reModding, 2015-05-28:
Yes. They do not beep after powering them and when I unplug the FC, they beep three times (power up). When I plug them to the FC now, they continue with two further beeps (throttle signal detected and zero throttle detected in PPM mode). At this time I can increase throttle and they spin up as expected.

How do I test if I got the same problem with the bootloader mode waiting for serial communication?

MarenB, 2015-05-28:
No reason to test any further, I also didn't do anything else. The phenomenon seems to be quite clear, at least for the developers :)

4712, 2015-05-29:
Please the latest BLHeliSuite https://github.com/4712/BLHeliSuite/releases/tag/13202
If you do a reflash with an ISP interface, it will update the bootloader and should solve this issue.

sskaug, 2015-06-04:
I believe this issue to be fixed with the 13202 release, so closing issue.

closed 2015-06-04 by sskaug

reModding, 2015-06-05:
Yes, it does. Thank's for support :-)

MarenB, 2015-06-07:
For me as well, thanks for the fast solution!