bitdump/BLHeli #71: contact

fabriziotappero, 2015-05-25:
Hi,
I am interested in the Blheli firmware and I'd like to ask for your opinion on some ESC stuff. Would you mind sending me an email at: fabrizio.tappero at gmail.com

thanks a lot
Fabrizio
PS I tried to find your contact over the internet but no luck

sskaug, 2015-05-25:
Done. Closing issue.

sskaug, 2015-05-25:
Hi,
This is my email .

From: Fabrizio 
Sent: Monday, May 25, 2015 7:37 PM
To: bitdump/BLHeli 
Subject: [BLHeli] contact (#71)

Hi,
I am interested in the Blheli firmware and I'd like to ask for your opinion on some ESC stuff. Would you mind sending me an email at: fabrizio.tappero at gmail.com

thanks a lot
Fabrizio
PS I tried to find your contact over the internet but no luck

closed 2015-05-25 by sskaug

imperanetgithub, 2021-09-22:
hello I'm also having a timing problem with esc blheli32 with 24p 330kv engine. Can someone help me

sskaug, 2021-09-22:
This is not the best place to discuss such problems. Can you post in https://www.rcgroups.com/forums/showthread.php?2864611-BLHeli_32-Power-to-perform?