bitdump/BLHeli #72: Blueseries 12A bad start

vratinov, 2015-05-25:
I'm running BLHeli 13.2 on BlueSeries 12A that I got from witespy. They have SimonK bootloader and BLheli 13.2 that I flashed.  Below are the settings.
[image]

I have an issue with 2 out of 4 motors on my quad, once it's powered up and the very first spin up, 2 motors surge for 3-4 times before they begin working normally. The motors are Emax MT1806-2280KV. Once they finish surging it seems they become in-sync and everything works perfectly. However each time I arm for the very first time with a new pack, this happens. The video can be seen here on youtube:
https://www.youtube.com/watch?v=ZwQsB-z2tB0

I tried recalibrating many times, this is not the case. 

This can be reproduced one of two ways:
Either by arming the quad on the first bootup or by bring up the slider in cleanflight to max (as in this video) at very fast speed. If I'm using the slider and I spool up slowly this issue is not visible. However this happens every single time I arm the quad after the battery had been changed.

Thank you so much,
Vadim

4712, 2015-05-25:
Set all settings to default values (beside of Throttle min /max, Motor Direction and PWM Frequency) and retry...

vratinov, 2015-05-26:
Results are the same. Do things like voltage limiter, beep strength, beacon strength and beacon delay can really make a difference? The default has Demag on low and I tried that, the results were identical.

sskaug, 2015-05-26:
Still working on improving the reliability of startup, and the results so far for the next rev are promising.

vratinov, 2015-05-26:
Anything I can try in the settings? It just bothers me that 2 always work and 2 always do the exact same thing. The ones that do that, have the direction set in reverse, but I did try setting it to normal and the results were the same. At first I thought the direction was actually the reason they did this. What causes these surges, could it be bad wiring to the motor? The motor is soldered directly to the ESC, but I did have to trim the motor wires in order to achieve this clean/short connections and this meant that I had to scrape off the lacquer of the shortened wires in order to tin them again. I did spend significant amount of time making sure they are tinned well but perhaps out of the 10 strands a few are not tinned, could this cause this issue? I mean I would like to eliminate all possible culprits. I do have brand new identical motors with tinned wires from the factory I can try these ESCs with and see if the issue continues.

sskaug, 2015-05-26:
Weird that it coincides with BEC. But then they are probably from different batches that could have an impact.
I have been doing changes to improve startup, if you want test code to evaluate, email me at sskaug at gmail dot com.

vratinov, 2015-05-26:
Got in touch with Mr. Sskaug who gave me a rev14 beta firmware. This firmware immediately fixed all my issues. Thank you so much sir!

closed 2015-05-26 by vratinov