bitdump/BLHeli #74: Better information about switching speed

nuclear12, 2015-05-30:
Hello
I'd like to get a better grasp on the descriptor used for the fet switching speed
currently, i've seen:
switching speed is fast with a value (like <0.5us)
switching speed is not fast with a value (like 5us)
switching speed is slow
switching speed is quite slow
switching speed is quite fast
switching speed is somewhat fast

would it be possible to get an order like ?
quite slow
not fast
slow
somewhat fast
quite fast
fast

Thanks

samlawyy, 2015-06-01:
yes, I just confusing about quite fast and fast, which one is the faster?

sskaug, 2015-06-02:
Sorry guys, I know this is a bit confusing and not very precise. But atm I do not have the capacity to detail this further.

closed 2015-07-08 by sskaug