bitdump/BLHeli #76: Can't ID Atmel ESC

CradyCopter, 2015-06-04:
Hi, I have some Atmel-based ESCs that I'd like to flash with BLHeli but I can't match them to any of the images in the BLHeli supported Atmel ESCs.pdf

Does anyone recognize this PCB and/or can someone please tell me which pads to use for programming with a BLHeli Box?  I've tried contacting Ready To Fly Quads but their support is nonexistent — they never answer their emails.

Thanks!

[image]

sim-, 2015-06-04:
Looks to me like bs_nfet :)

4712, 2015-06-04:
Have a look here: http://www.rcgroups.com/forums/showthread.php?t=2305535&page=174#post31524345
and here http://www.rcgroups.com/forums/showpost.php?p=31528855&postcount=2604

May be this helps.

CradyCopter, 2015-06-04:
Thanks for the quick responses, guys!  

4712, those are exactly what I have -- thank you!  BTW, I have a Turnigy USB Linker, do you know if it's possible to flash these ESCs with BLHeli using the Linker, or will I have to use the box?

4712, 2015-06-04:
No, simply use the Turnigy USB Linker.

CradyCopter, 2015-06-04:
Cool!  I'll try that this afternoon.

If I run into a snag or can't find the docs I need can I run to you for 
help?

=P

CradyCopter, 2015-06-04:
The Linker worked brilliantly, thanks!

Selected BS 30A and seems to be working good.

=)

closed 2015-07-08 by sskaug