bitdump/BLHeli #80: Support for SL C8051F85?

kriswiner, 2015-07-09:
First thanks so much for creating this site and the BLE suite GUI! You have made life very easy for us. 

We redesigned the MX-3A ESC using the C8051F85 chip (40% smaller, internal 1.65 V reference). We successfully used the GUI to reflash a stock MX-3A with the F330 and want to repeat the process with our custom F85 board (it is basically a copy of the MX-3A circuit but with the F85, an on-board 3V3 voltage regulator for the MCU, and TI's femtofets). I don't think the F85 requires the voltage divider reference since it has its own internal 1.65 V reference but in the first iteration I just copied the MX-3A circuit. 

I know the F85 C51 register map is almost identical to that of the F330 except for a few differences. I wonder if it would be possible (and how?) to use the GUI to program our custom ESCs by making these changes. Can we modify and compile (using Simplicity Studio) the files on the github repository to create our own hex files to flash with the SL toolkit, or can we modify the GUI to respect the F85 variations using the MX-3A ESC configuration. Can you suggest a path forward for us?

Thanks for your help!

Kris
www.flynixie.com

4712, 2015-07-09:
BLHeliSuite will hopefully be able to flash the F85 in rev14 (soon).
What is the exact kind of F85 you are using (DerivativeID)?

kriswiner, 2015-07-10:
Glad to hear it's in the works. I bought ten of the C8051F850-C-GM 8 kB/512B
RAM chips from Mouser (their part number 634-C8051F850-C-GM). I can always
get a different kind of F85. I have no real preference as long as it works
for us.

Kris

<https://github.com/bitdump/BLHeli/issues/80#issuecomment-120163118> .
<https://github.com/notifications/beacon/AGY1qgffMV8lVukbXtaZLlNAEneJwYp1ks5
obumTgaJpZM4FVpoI.gif>

sskaug, 2015-07-10:
The 'f850 should work fine. If/when you have some HW, we'll be happy to
look at it and make code for it.

kriswiner, 2015-07-10:
Excellent and thank you! I have an EAGLE design package which has a schematic.

https://www.oshpark.com/shared_projects/BNvKHcQZ

Is this what you mean or would you like samples of the assembled hardware? I will thicken the traces to the motors in the next iteration but this design is sufficient to check function.

I did rearrange the pin assignments to make it easier to layout the traces so some changes in the coding will be required. I suppose I could make an exact copy of the MX-3A to save the trouble but I would like the flexibility (and understanding!) to be able to adjust the firmware to the hardware design rather than vice versa.

I wonder if you can comment on the purpose of the voltage divider on pins P1.6 and P1.7 (IIRC) on the F330 of the MX-3A. Is this to provide a 1.65 - 2V voltage reference for the PWM comparator function or something else. I am hoping I can get rid of it when using the F850 but until I understand its purpose I can't.

I greatly appreciate your support. When I get the 850 ESCs to work I plan to make them available for sale at Tindie as well as use them in our commercial product. I am happy to provide you with samples for your own testing.

Kris
www.flynixie.com
https://www.tindie.com/stores/onehorse/

4712, 2015-07-12:
BLHeliSuite 14.0.0.0 is released now. Flashing F850 is tested and should work in all aspects (C2 and BLHeli bootloader).

sskaug, 2015-07-12:
The board looks really small and cute!
The voltage divider on the P1.6 input is for low voltage limiting - reducing motor power when battery voltage drops. And in order to create code, physical HW is required :).

But - for the Nixie, don't you have to think beyond this?
Wire bond an 'f850 and wire bond bare die fets to flex cables? Absolute minimum weight.

kriswiner, 2015-07-12:
I would be delighted to send you a couple of the first ESC boards for your code development and testing. I expect to get the pc boards from the fab next week. Please tell me where to send them!

In the mean time I have changed the FETs to TI's NexFets which have very small on-resistances; 16.2 mOhm for the P-type FET, which is incredibly small. I am hoping this ESC proves to be very efficient. Here is the latest board design; it keeps the pinout of the first one:

https://www.oshpark.com/shared_projects/fghBGBtS

You can see in the EAGLE schematic I have reassigned the  gates and low-voltage detection circuitry as well as the feedback input to different pins than the MX-3A uses. I hope this doesn't cause too much trouble.

Is there somewhere an explanation of how this voltage divider works to limit the motor power when the battery voltage drops or is this just part of the coding? Also, can you help me understand the purpose of both a pullup and a pull down on the incoming PWM signal line? Only one is usually populated, of course, but why are these necessary in the first place?

Currently, in our Nixie prototyping, we are using MX-3A ESCs which ride outboard and under the BLDC motors. We want to bring the ESC circuitry inside the main compartment since we want to minimize motor height, which dominates the thickness (or thinness) we can achieve. I don't want to integrate the ESCs onto our main board yet since replacing a FET there would be very cumbersome. Rather, I want a thin, small footprint but separate ESC board that can easily be replaced when damaged (this happens a lot!).At least in the short term. In the longer term I am exploring whether four ESCs can be replaced by a single 32-bit ARM processor and twenty four FETs. If so, then this might make it onto the main board in the commercial product. Even if not, or if there is some other drawback to this approach, we might just include four F850s and the corresponding FETs on the main board in our commercial product.  

I think for the general RC community, these smaller, more-efficient ESCs might be very popular, especially if you are able to write appropriate firmware and we are able to tune it for popular (and small) BLDC motors like those for sale at HobbyKing.

I am looking forward to using these ESCs with the BLHeli firmware! Thanks so much for your interest and support.

Kris

sskaug, 2015-07-13:
I'd be happy to test HW when available, and develop code.
As for your design files, I don't have SW installed to read neither the schematic nor gerber, so it's a bit hard to comment on the design. How about pdf files? 
Are you using minimum size resistors and caps (0402 or smaller?). Maybe you can even reduce size below the 0.4x0.4? Those fets on the backside are really tiny.

Reassigning pins is generally not a problem, but there are some limitations in the MCU HW.

The low voltage limiting is done by reading the (divided down) battery voltage by the ADC, and then limiting motor power in FW.

Ping me at sskaug69 at gmail dot com, and we can exchange address

sskaug, 2015-07-18:
The Fairy code for its 'f850 is working and released, so BLHeli now supports the 'f850 MCU.
Closing this issue for now.

closed 2015-07-18 by sskaug