bitdump/BLHeli #82: EMAX 12A Simon Series with v14 problems!

arrowcircle, 2015-07-14:
Hi! Today I flashed my ESCs with BLHELI 14.
After calibration and cc3d reconfiguration I came to flying and got a crash.
When flying, when I set zero throttle and increased it back my quad flipped forward and crashed.
I tried again and got a crash again.

I developed what happened. After getting zero throttle and increasing throttle back motors start at different time.

What I doing wrong? Is my ESC incompatible with v14? If so, please be aware.

peisel, 2015-07-17:
I had tested the EMAX 12a with BLHeli 14.0 with this ESC and DYS 1806 ... sounds bad and getting hot ... best flyable setup was 13.1 with medium high setting.

sskaug, 2015-07-18:
The EMAX ESCs do not have the best of quality when it comes to matching of the bemf sensing network. Therefore, they can stutter a bit when starting. When flying you should always have the motors running (idling, even with zero throttle), to my knowledge that is the general setup of an FC.

As for the DYS 1806 2300kV and EMAX 12A and 4S and 5030, my test setup runs just fine.
Mid throttle at 2.5A and full throttle at 9A

arrowcircle, 2015-07-20:
Could You please add 13.1 as max version for emax 12a in docs? And add, that v14 and higher are not usable for these esc.

peisel, 2015-07-20:
SSKAUG would you please share your settings for blheli.

sskaug, 2015-07-20:
Just defaults, except for damped light :)

closed 2015-09-16 by sskaug