bitdump/BLHeli #83: Issues with UBAD20A and Cobra 2204/1960

Javaello, 2015-07-20:
Today I had some weird occurrences with my Cobra 2204/1960kv and Ubad 20A ESC's. With BLHeli 13.2, it was no problem to run 6x4.5 props and 4S. Today, after flashing this morning before flying to BLHeli 14, I started having problems. Startup and run through 3/4 throttle, it would run great, push it to full throttle and the motors would drop out and the quad would fall out of the sky. Running it on 3 or 4S and 5x4.5 props, it was flying great. After talking with one of my local pilots, he was having great luck, until he ran 6" props. Attached is my screenshot from BLHeli Suite.
[image]

sskaug, 2015-07-20:
I just ran this ESC and motor with a 6x4.5 prop on 4S on the bench, with the above settings.
Both with 13.2 and 14.0, and the most noticeable difference between 13.2 and 14.0 is the improved braking on 14.0.
This motor is well behaved when it comes to demag, I could not observe any hints of sync loss.
Also, max current was pretty similar, some 18ish amps.

Not sure what the issue is in your system, are you sure your ground and power routing is good?
Not that I fully understand why this should cause a difference between 13.2 and 14.0, though. But maybe the increased braking increases the dynamic current consumption? In general, I think the P-gain and maybe also the D-gain can be reduced due to the improved braking.

Javaello, 2015-07-22:
I have solid grounds and the ESC leads still have their ground. A couple of things I remembered, on my last crash, I did a flip which I reduced power prior to. I had start up issues in flight and could not recover from it and fell from the sky. Should I increase my startup power or change demag comp to make this more consistent? I wish I understood demag a little better. Can you elaborate on what you mean by "This motor is well behaved when it comes to demag"? Should I turn it to low, high, or off?

sskaug, 2015-07-24:
I mean that this motor has fast demagnetization of the windings, so sync loss due to too long demag time is generally not an issue. It runs very well with default settings.
Don't you set up your FC so that motors are always idling?

closed 2015-09-16 by sskaug

light24bulbs, 2016-01-14:
+1 I am having the same exact issue with the 1960s.  I have found demag compensation is a good fix but I'm curious if it is the RIGHT fix, given that it decreases motor power.  @Javaello did you find a good solution in your settings?