bitdump/BLHeli #86: ESCs are not arming when provided with a higher frequency pulse sequence

johnwthomson, 2015-08-14:
I'm working to drive my ESCs (found here: http://www.readymaderc.com/store/index.php?main_page=product_info&cPath=505_507_658&products_id=4123) using OneShot125 at an update rate of ~1KHz +/- up to 500uS of jitter.  At 800Hz, everything seems to work properly, but when the frequency is upped to match the sample rate of my gyros (1kHz), the ESCs seem to not be detecting the pulse train.

I have tried configuring the ESCs to be OneShot125 only (no PWM support) to eliminate the possibility that they are detecting some ambiguity in the pulse train, but that has not made any difference.  The "Power On" tone is played, but the "PWM signal detected" tone is not.

Does BLHeli support a 1kHz pulse train in OneShot125?  How much jitter is permissible at this frequency?

sskaug, 2015-08-15:
I've had a look at this, and have a code that I think can fix the issue.
Ping me at sskaug69 at gmail dot com if you want to try.

johnwthomson, 2015-08-16:
Issue resolved.  1kHz pulse rate now accepted on boot in tests.  
Fix will be publicly released in v 14.1, according to sskaug.

closed 2015-08-16 by johnwthomson