bitdump/BLHeli #88: Closed Loop not getting to sufficent speed

Thor368, 2015-08-23:
Hi there!
I have an issue with closed loop mode in multi firmware.

Background: I recently switched over to much bigger props (from 8 to 12inch) and figured that the quad is unflyable right now due to high interia of the new props. Using SimonK on Hobbyking F20As back then I started reflashing the ESCs with BLHeli to get Dampended Light and Closed Loop control. Dampend Light is working fine for now (besides the full stop breaking at 0 PPM being a bit to short so that the props are still spinning at low speed when motor break is released).

Now the Problem: Closed Loop Mode is driving me nuts. First up the feature seems to work as supposed. When the motor is spinning and I apply force to it the ESCs is fighting it until stall followed by restart. The funny thing is that the motor is spinning ridiculously slow. Maybe one-tenth of max RPM without Closed Loop even at full PPM (1850us in my case). Above 1150us I get no change in RPM whatsoever.

Things I have trided to get rid of the low RPM issue:
- recallibrating throttle input -> all fine without Closed Loop
- trying out all Closed Loop options from LoRange up to HighRange -> no change in max RPM
- using OneShot instead of PPM
- using High and Low PWM Frequency instead of Dampned Light
- playing around with Closed Loop gains

This problem is the same for all four motor ESC combos. What could this be?
Thor

joshuabardwell, 2015-08-23:
On an unrelated note, I'm surprised you can't fly in normal mode with 12" props. Lots of people are flying with 12" props or even bigger, without enabling closed loop mode. Are you sure something else isn't wrong?

Thor368, 2015-08-23:
Well, "unflyable" is ment in an acrobatic sense. Sure enough if i dial down the PIDs a LOT things start to get controlable but that is not what I would call a pleasure to fly in loopings^^ Main cause for this is mostly inertia induced lag. Especially problematic is that spin up and spin down times are asymetric. Dampend Light and Closed Loop mode would be a huge improvment if working correctly.

joshuabardwell, 2015-08-23:
The effectiveness of damped light depends largely on the hardware in your ESCs. The firmware matters too, as we can see by comparing BLHeli 13 to BLHeli 14, but the hardware limits the maximum braking that will occur.

I'm not sure I understand how closed-loop mode will improve your spin-down time.

Honestly, there is only so much you will be able to do, acrobatics-wise, with 12" props. There's a reason most acro quads use smaller props.

Thor368, 2015-08-23:
Thanks for your thoughts but it was not the personal opinion on how I should fly my quadcopter that started this thread. It would be nice to hear something about the low max RPM of BLHeli.

PS:
There is a realy nice piece of work showing the difference between Dampened Light with and without Closed Loop mode.
http://www.rcgroups.com/forums/showatt.php?attachmentid=5550610

sskaug, 2015-08-23:
Have you tried older code revs?
There have been significant changes to the code, and I haven't done much verification on closed loop.
Maybe a bug has sneaked in...

Thor368, 2015-08-23:
OK, I will try that when sun is up again^^.

Thor368, 2015-08-24:
OK in 13.2 the max RPM is working as expected. Although the controller performance is pretty poor. There are a few things to improve. I am somewhat into AVR programming is there a way to contribute?

Thor368, 2015-08-24:
Ah and disabling low RPM power limit didn't do anything for me.

sskaug, 2015-09-16:
Fixes implemented in rev 14.1

closed 2015-09-16 by sskaug