bitdump/BLHeli #89: Motor reversing is slow

codepox, 2015-08-29:
Hello sskaug,

I have been doing more testing and took measurements of the motor reversing.

I am using Blackbox to record:
- Throttle Position 
- Battery Voltage
- Battery Current

The first ESC is the Afro 30A OPTO running SimonK 4/19/2015 with [comp_pwm] and [rc_puls_reverse] everything else default.
The second is the Rotorgeeks RG20A running BLHeli 14 with "Bidirectionnal" and "Damped light" everything else default.

The setup.
Battery: Glacier 4S 1800mah
Motor: Sunnysky 2212 1400kV
Propeller: Graupner 8x4.5" 3D
Flight controller: SPRacingF3 running Cleanflight 1.9 with "Oneshot125" and loop time = 1000
Receiver: FrSky X4R-SB
Transmitter: Taranis.

To have a fair comparison, I programmed my Taranis to send the sequence of pulses automatically, so there is no manual intervention.
On the graphs, the voltage and current amplitude have no meaning because the blackbox viewer use different scale values for every files but the time scale is the same (one point every 1ms, so 1KHz sampling rate).

The first par of the graph shows quick acceleration and deceleration, the second part shows the motor reversal.
For each I took a snapshot at zoom 10% and zoom 100%.

A few things looking at the graphs:
1 - The Afro 30A with SimonK runs smoother, a lot less noise on both voltage and current
2 - The Afro 30A with SimonK reverse significantly faster, twice as fast.
3 - The RG20A with BLHeli has a latency of ~150ms before anything happen

[image]
[image]
[image]
[image]

sskaug, 2015-08-30:
It sure appears that SimonK is faster. I have it on my list to also take some data.
In your plots, the initial time where no current flows is not a period where nothing happens. The motor is braked to a stop. And while braking, the motor does not consume any power.

codepox, 2015-08-30:
OK, that makes sense.
Do you have to break? Seems like the SimonK software start driving in the opposite direction right away.

sim-, 2015-08-30:
Yeah, I have no special handling of direction change at all -- it just starts to commutate against rotation, and the zero crossing timeout is calculated in such a way (not originally intentional) that allows the motor to not be considered stalled or out of sync as it spins the wrong way. I've been meaning to improve this for a while, too, since I've had a few requests to do it with braking instead, and it doesn't even change direction until the commutation loop starts over (it only checks direction every 6 commutations). Also, as the motor crosses 0, it may or may not time out and cause a fall back to start code which imposes a duty acceleration ramp. Probably Steffen's method is more repeatable? This is important, for example, if you want to have equal thrust from each motor over the time of the direction change.

codepox, 2015-08-31:
Hi Simon, 
What ever you do or don't do actually works quite well.
I have done many tests using both Silabs and Atmel ESC running BLHeli and SimonK software and I always had the best results with Atmel ESC running SimonK, especially the ones using ICP1 like the Afro HV 20a, Afro 30a and SN20a.
The reversal is fast and very predictable.
I have some videos on my youtube channel if you want to check them: https://www.youtube.com/channel/UCCEzU3gvkrl-VckRk4hPE2A
Also, I started a thread on RCGroups dedicated to 3D flying.
http://www.rcgroups.com/forums/showthread.php?t=2480732
If you are making improvements to your 3D code and you are looking for somebody to test, please contact me.

I have a question I always wanted to ask you, is there a particular reason why the 3D neutral point is at 1460 and not 1500?

sim-, 2015-08-31:
The endpoints in my case are purely historical. They're actually set based on what lined up to my particular servo tester on an old Turnigy Plush with internal oscillator, which doesn't quite line up to 16Mhz. Now it's sort of expected by some people. It might make sense to change it with a flag day at some point, but then I can never decide what's best to set it to. Anyway, 1460µs is half-way between 1060µs and 1860µs, my default endpoints. There's also a default dead-band of 100µs (50µs in each direction) that is turned on with RC_PULS_REVERSE. This and the 3-point calibration code was all set when I tried it in my old RC-10 with AM radio (figured that was about worst case...).

codepox, 2015-08-31:
Don't change it, I was just curious. Cleanflight uses 1460 as the default value for 3d_neutral.
You are talking about calibration, should I use it?
So far, I have just set these values in Cleanflight:
min_thottle = 1060
max_throttle = 1860
3d_neutral = 1460

jamesdeep, 2015-09-23:
Hello,codepox,
May I ask what kind of tools you used can draw these plots?

codepox, 2015-09-23:
I use the Blackbox capability of Cleanflight software running on an SPRacingF3 flight controller board.
The SPRacingF3 has analog inputs for battery voltage and current as well as an SPI Flash for onboard data storage. I use an external 90A high side current sensor.
The graphs are rendered by the BLackbox viewer app.

I recently too measurement on a KISS 30A ESC:
[image]

sskaug, 2015-10-28:
Rev14.2 is now out, with very fast reversals!

closed 2015-10-28 by sskaug