bitdump/BLHeli #90: "Jittery" flight after upgrade to BLHeli 14

joshuabardwell, 2015-08-30:
Before upgrading to BLHeli 14, I had what I felt was a pretty good tune. An example of flight from that tune can be seen [here](https://www.youtube.com/watch?v=PmxE3mHYYu0). This tune isn't perfect--P is a little too high, and/or TPA is needed on high throttle climb-outs--but compare it to [this one](https://www.youtube.com/watch?v=jSR73T8T1yw), from today. Notice the "jitters" that are sort of constantly present throughout the whole flight. These showed up when I upgraded from BLHeli 13 to BLHeli 14, and I simply cannot get rid of them no matter what I do. I have decreased P to the point where I was unable to reliably hit tight lines and crashed a lot. I have even filtered down to a pretty extreme level. [This example](https://www.youtube.com/watch?v=EIFk6aJFb0E) has gyro_cut_hz at 50, pterm_cut_hz at 40, and dterm_cut_hz at 6. I have also tried pterm_cut_hz at 35 and gyro_cut_hz at 40, although I don't have a flight video of it. Filtering more heavily reduces the problem, but it's still there. And I was flying on gyro_cut_hz=70, pterm_cut_hz=50, and dterm_cut_hz=15 with no problem on BLHeli 13.

I'm not sure what could be causing this. Could it maybe be dither? I don't know... I don't think it is a reflection on my ability to tune, and I have been completely unable to get rid of it, so I'm submitting it as a bug just for the record.

The file below is renamed to PNG so that Github will accept it, but it's actually a Blackbox log from today's last flight (example 3, linked above).

[image]

sskaug, 2015-08-30:
I don't hink the dither is the cause.
Are you using a typical Atmel ESC, like the SN20A?
If so, then I think a likely cause is the improved braking.

joshuabardwell, 2015-08-30:
I'm using F-12A, which is Atmel. What's your recommendation then? Go back to 13.1 or 13.2?

closed 2015-09-01 by joshuabardwell