bitdump/BLHeli #91: Can't add external ESC .Hex files

ckalpha, 2015-09-05:
Just put Vendor specific versions of EMAX 12A BLHeli into BLHeliSuite sub directory called  "BLHeli_HexFiles\Silabs", How actually to activate new .hex files to ESC.cfg ?

Thanks a lot

4712, 2015-09-05:
I had no chance to test an EMAX ESC yet.
AFAIK, they are yet not using an official supported bootloader.
Also the file structure is different from the standard. So, BLHeliSuite is not able to recognize it.
You can only flash a unknown SiLabs file using the a C2 capable interface and pick the file manually...

Perhaps you should contact the vendor support and ask about the best way...

ckalpha, 2015-09-05:
May i can even force flash  .hex via BLHeliSuite expert tools manually ?

4712, 2015-09-05:
Yes. Pick "Flash any file".

ckalpha, 2015-09-05:
Just manually flash with EMAX specific .hex, work like charms! but still has no chance to get setting information out.

Thanks a lot.

closed 2015-09-14 by ckalpha