bitdump/BLHeli #92: Stuttering and stalling during acceleration

grubsauce, 2015-09-11:
Hello,

I am using Arduino to supply a pulse width to a 30A Afro ESC to control a 650 kv BLDC motor that is connected to a steel flywheel. The major goal of this project is to generate torque. I am using BLHeli 14 with full power at startup, medium high timing, and bidirectional enabled.

From a standstill, the largest step in pulse width I can take is from 1488 (neutral) to 1650. Any step greater than that causes the motor to immediately stutter/stall (I assume this is a loss of sync?). While the motor is spinning at its lowest speed (1515 micro second pulse width), I expect to be able to send full power (1860 micro seconds pulse width), however, the most I can send without stalling is 1700 micro seconds.

I understand the inherent limitations of a sensorless setup, but I expected the performance (of a sensorless setup) to be a bit better. I assume the behavior I am experiencing is normal for this type of setup but I do have a couple questions.

Does anyone have an explanation for the behavior I am experiencing? If I understand the source of the stuttering/stalling, I can strategically add delays between pulse width commands. 

Is there any way to modify the Afro ESC to accept hall sensors as an input? Or maybe to trick the ESC into using the hall sensor signals instead of the back emf? I will be installing hall sensors in the appropriate locations just in case I need to move to a full sensored ESC in the future.

Thanks for your help.

sskaug, 2015-09-11:
Well sensorless systems rely on bemf to sense position. And at low rpms, the bemf voltage gets small, while motor currents can get really huge. So it's kind of a doubled challenge to read bemf signal in the presence of noise from motor currents.

What can help in BLHeli, is to set demag comp to high. And possibly also to reduce startup power and/or increase timing.

Hall sensor inputs is a whole different setup. Hardware must be modified (of course) and also significant changes must be done to FW. BLHeli currently has no support for it, and there are none planned either.

joshuabardwell, 2015-09-11:
Would low rpm power reduction also help?

sim-, 2015-09-11:
Hello! Yes, this can be tricky particularly when there is DC bias offsetting the timing sensing. It might be worth comparing with other firmware as well -- I have compensation for this (off by default) as well as some ZC filtering "limited backtracking" at lower speeds, and a light duty cycle limit calculated from the timing. You could also try lowering PWM frequency. However it is accomplished, it is important to make this case work reliably, because it happens also in multi-rotor flight like when yawing one way quickly to the other.

You could also try an RC car ESC with hall effect sensor input, if all else fails. :)

grubsauce, 2015-09-14:
Thanks everyone for the help and the quick responses! I set demag comp to high and that made an incredible improvement in the stuttering/stalling. I can now go from the lowest pw (1515) to the highest pw (1832) without any stuttering. Additionally, I can go from max pw (1832-> 9200 rpm) to 1370 pw in the opposite direction without any stalling issues! Sometimes there is a delay around the zero-crossing but that probably has to do with the sensorless. I also recorded a lot of measurements and found high timing to give the best performance so far. 

I would love to test the SimonK firmware to see how it compares to BLHeli but I just can't figure out how to modify the firmware in kkmulticopter.

I have just a few more questions. For BLHeli 14, does startup power effect the zero-crossing? Or does it only effect things when the motor is first started?

I know BLHeli doesn't support hall sensors, but is it physically possible to give the AFRO ESC, i think it has an atmel in it, sensor inputs?

Just in case you're interested, here's a link to an article about high performance sensorless control of BLDC motors.

http://www.ti.com.cn/cn/lit/an/sprt703/sprt703.pdf

Thanks for all your help.

sskaug, 2015-09-14:
Great to hear that it helped :).
Startup power will have an effect on direction reversals with BLHeli.
And thanks for the pointer to the article, looks like interesting reading.

closed 2015-09-16 by sskaug