bitdump/BLHeli #94: 14.x - RCTimer hvsk 40a reboots on the very first throttle input

adfolfotregosa, 2015-09-19:
Hello. I have a strange problem with these esc after version 13.2.

On the very first throttle input the esc reboots. I have 3 esc with this symptom and another 3 that have yet to shown it, they work first time.
The motor plays the startup tone and the input signal detected tone. You give a litle throttle and it reboots. It does the startup tones all over again and then it works perfectly. A video is best to show what I mean. 

https://www.youtube.com/watch?v=vKqXwJU-Os0

Downgrading back to 13.2, after the signal detected tone plays it works first time. I presume this is not hardware error.

https://www.youtube.com/watch?v=3O8BAftS0ws

Thank you for your time on this.

joshuabardwell, 2015-09-19:
It would probably be useful to know your settings. Specifically, whether you are using OneShot input, and whether you have PWM input disabled on 14.

adfolfotregosa, 2015-09-19:
Of course. Sorry.

13.2 settings:
https://drive.google.com/file/d/0B7E54-YVLWHaZFQ5MkZtYVpPa2s/view?usp=docslist_api

14.1:
https://drive.google.com/file/d/0B7E54-YVLWHaZG5mdW5nTzdWVzA/view?usp=docslist_api

sskaug, 2015-09-21:
Thanks for the feedback.
I have now dug up my old hvsk40A, and I can reproduce what you see.
Still need to dig further to understand it, but for some reason the MCU resets the first time I enable the ADC. Weird.

adfolfotregosa, 2015-09-21:
Well like I said, I have some that work first time and others that reboot on 14.x 
On 13.2 they work correctly. 
If you need me do some firmware testing just ask. 
Thank you for your time.

sskaug, 2015-09-26:
This has to do with the vref pin that has a pretty large decoupling cap, and when that cap is charged (when enabling the ADC) the brown out detection of the MCU gets triggered. It is a weakness of the hardware design, ESCs like Afros have the same structure, but do not have the problem.

I think the code for this ESC can be modified somewhat, to not show the problem any more. If you ping me at sskaug69 at gmail dot com, I can send you test code.

sim-, 2015-09-28:
Hi sskaug: You mean the capacitor from AREF to ground? The datasheet specifies this as the only supported configuration for the ADMUX REFs bits ("AVcc with external capacitor at AREF pin") that allows AREF to be 5V without bridging it externally (should have better noise than by bridging it, and this also supports 2.56V internal option). They don't seem to specify a recommended capacitor size, but Afro has 100nF. I guess you could momentarily drive it to 2.56V, which would work as long as it's not accidentally bridged on some boards, but I'm surprised that charging that cap alone would cause a  brown-out! I've tested that pin in the past (abusing it as an output) and it is fairly high impedance.

sskaug, 2015-09-29:
Yes, it's a cap on the AREF pin. And charging it (to 2.56V when enabling the ADC) causes brownout. Weird. I'll look more into it when I have time.

sskaug, 2015-09-30:
Not always too fond of the Atmega8A...
On this ESC, the BOD is enabled (probably is on most), and this uses the internal 2.56V reference.
So, the reference is always on. But only when enabling the ADC (with REFS0 and REFS1 set), is the reference connected to the AREF pin. And if AREF is not charged, it takes some time to charge it (in this case about 8us), during which the reference is pulled low, and we have a brown-out.

sim-, 2015-09-30:
Wouldn't pulling the reference low cause the opposite? (No brown-out if there should be one.) Unless AREF was already higher.

sskaug, 2015-09-30:
Of course you are right - my thinking got a bit crossed :).
But then again strange things can probably happen if the reference is very low (below the thresholds of the fets used on the chip).

sskaug, 2015-10-28:
Fixed in rev14.2.

closed 2015-10-28 by sskaug