bitdump/BLHeli #95: No bootloader support for opto esc

0n3-70uch, 2015-09-21:
Hello,

I have a jeti controller with an optocoupler at the RC input pin (inverted signal). I did my own blheli configuration (.inc) for this atmel ESC. Everything works well, but the bootloader does not respond. I assembled the code with AVR Studio 4.
I have tested the compiled code with a RCtimer 45A NFS ESC. On this controller the bootloader works well. But the RCtimer ESC has no optocoupler.

The jeti controller works with an external 12MHz quartz. So I add the new cpu frequency as a compiler option. Is it a problem for the BLheli software?

I hope you can help me.
Greetings from Germany and sorry for my bad english 
Fabian

tracernz, 2015-09-21:
If it has an actual opto-coupler (likely it does since Jeti =/= China) it will be uni-directional, so there is no way for a signal to travel back from the ESC.

0n3-70uch, 2015-09-21:
damned.... :-)

Thanks Michael for your fast answere. Working for more than five hours on the same problem. And the answere is so easy and logical ;-). 

Tomorrow I will test a bootloader with an other pinout. No great solution but better than nothing.

I am searching for a high quality BLHeli ESC. The JETI controllers are great but the optocoupler and the 12MHz quartz are not so fine. Have you an idea for a very good ESC?

sskaug, 2015-09-26:
The XRotor series are very good ESCs.

0n3-70uch, 2015-09-26:
Very funny. Yesterday I bought two Xrotor pro 40A escs. Nice quality but no blheli suport. The prozessor looks like a arm cortex cpu. Is it right? I attached a picture.

[image]

sskaug, 2015-09-26:
Cool, thank you for the pic.
Yes, it appears the 40A Pro version uses a 32bit ARM MCU. And so does the XRotor 15A. It appears that Hobbywing are moving over to a 32bit ARM platform, at least for some of the ESCs. ARM MCUs are not supported by BLHeli. But AFAIK, these ESCs come with good factory FW, that runs "damped light" (HW has another name for it).

0n3-70uch, 2015-09-26:
Not at all! I tried to open the other side. But this is bonded to the heat sink.
The factory firmware is really nice, everything works well. So I use the ESCs with the original firmware.

I think Hobbywing calls "damped light" > Deo (Driving Efficiency Optimization). Is there any plan to convert BLHeli for ARM MCUs? I can send you an ESC!?

Greetings
Fabian

sskaug, 2015-09-26:
Good to hear that it works well.
As for BLHeli on ARM, that's a considerable effort!
And not too many escs out there, at least yet.
So at the moment no plans.

jamesdeep, 2015-09-26:
Hello sskaug,
May I ask what does "BL"mean ?It confuse me for quite a while :)

sskaug, 2015-09-26:
Well, it started with the mcpx. Making a decent BrushLess setup for that heli :)

Trex4Git, 2015-09-26:
As that would be a different µC, especially the low level assembler code must be ported to a new aritecture,  I think.
But on the 32bit µC maybe not more assembler programming would be absolutely needed. And C programming might be easier as assembler.

closed 2015-10-06 by sskaug