bitdump/BLHeli #96: [Feature Request] Regenerative Breaking

trashmaster9000, 2015-09-25:
Hi,

are there plans to implement Regenerative Breaking? (Copter/Main?)

I think its usefull especially for large ESCs (eg. K-Force v2 HV, i use this in my E-Bike)

It could also be usefull for Copters with big Props.

joshuabardwell, 2015-09-25:
Supported since at least version 11.0

http://www.unmannedtech.co.uk/uploads/6/7/0/2/6702064/blhelimanualsilabsrev11.x.pdf

trashmaster9000, 2015-09-27:
I cant activate Damped Light in my K-Force 120A v2 HV ESC.
Why is it not supported in this ESC?

joshuabardwell, 2015-09-27:
Usually, the answer is that the FETs don't allow it.

trashmaster9000, 2015-09-27:
Ok. but what is limiting?
The FETs, the CPU...?

Can i activate and test it on my own risk?
I think i just Need to Change DAMPED_MODE_ENABLE to 1?

joshuabardwell, 2015-09-27:
I don't know for sure. It has something to do with the FETs. I have heard it said that braking performance has a lot to do with how fast the high-side drivers are. If you don't have fast high-side drivers, you won't get good braking, even if the ESC is trying to brake. 

I would not advise turning it on if @sskaug has decided it should be off. He probably has a good reason. But if you're willing to toast the ESC and/or possibly risk a fire, go for it.

sskaug, 2015-09-27:
Damped mode was disabled for this ESC, as it was hard for me to verify that it is safe. 10S and 120A is more than I can test here. That being said, it should be capable of running damped light.

Just set:
DAMPED_MODE_ENABLE to 1 in the inc file, and 
DEFAULT_PGM_MULTI_PWM_FREQ to 3 in the main asm file,
and compile.

joshuabardwell, 2015-09-27:
Speak of the devil... :-)

trashmaster9000, 2015-09-27:
Ok. i will test it...
What could go wrong ;D

joshuabardwell, 2015-09-28:
Probably nothing. Maybe fire.

closed 2015-10-28 by sskaug

Engineereric, 2018-07-13:
I think there is a mix up here. What BLHeli does is "active breaking" where by phases are effectively shorted to dissipate rotational energy. True regenerative breaking needs the back EMF of the motor to be higher than the battery and for this the motor needs to be over-spun such as during an auto-rotation in a heli.
In my case I want to use a brushless motor as a starter for a 170cc internal combustion engine and as the starter does not use a freewheel when the motor starts it will spin the brushless much faster than it would when driven by the battery and so I want BLHeli to commutate the phases and apply the DC back to the battery.
In this case it is a 3000W motor and we plan to pull 2.5kW off the generator. It works with diodes but I would like to have it all work with BLHeli and not just use the substrate diodes in the MOSFETs.
Thoughts?

sim-, 2018-07-13:
Shorting the phases does not just dissipate energy -- it also causes a flyback / boost effect, just like a boost converter. This very much results in the phase voltages being higher than the battery voltage. You should find that the commanded duty cycle will in fact cause a mostly proportional regeneration of current back into the battery.

In the case that the motor is already spinning fast enough to cause back-EMF higher than the battery voltage, all an ESC can do is track it to avoid FET body diode forward voltage losses. You won't be able to "open" the circuit.

Look on the 'scope and see if the high side (or low side) is pivoting around one of the battery legs or if both the high and low sides are a diode forward voltage away from the battery voltage. Remember that 100% duty is equivalent to tracking the motor and always keeping the high and low side FETs on, which is the same as having no forward voltage losses. You will get trapezoidal torque ripple, however. If, instead, your engine runs the motor slower than the maximum speed that the electric motor would spin it at 100% duty, you can select a duty somewhere in the middle to meet your regeneration target.

In terms of maximized efficiency and minimized torque ripple, it gets a bit more complicated.. :)

xnoreq, 2020-01-09:
@sim- I don't see how BLHeli(_S) does anything that would count as being regenerative. The opposite is the case.

When the phases are shorted then you won't have any boost effect and the current will choose the path of least resistance - the short through both phases' active high mosfets - instead of charging the battery.

You could get a regenerative effect by just having one phase's active high mosfet and the other phase's inactive low mosfet (after each PWM pulse).
Then, a voltage spike would build up and discharge through the battery, pulling up current through the low mosfet's body diode.