bitdump/BLHeli #97: Why was Stepped Start removed?

trashmaster9000, 2015-09-27:
Hi,

Why was Stepped Start with BLHeli 13.0 removed?
Wanted to test, if it helps to Start my E-Bike Hub Motor better. (i have to push him to start Spinning...)

joshuabardwell, 2015-09-27:
The latest versions of Suite have the ability to flash older hexes, but I'm not sure how far back it goes.

4712, 2015-09-27:
Back to rev 2.4 depending on the ESC...

trashmaster9000, 2015-09-27:
hmmm... where can i download older Versions of BLHeli?
Is there a Download Archive of BlHelisuite?
On the Homepage i can only download the latest Version.

joshuabardwell, 2015-09-27:
Download the latest version. Then flash the old firmware from there. The latest version contains an archive of the old firmwares all the way back to rev 2.4, as @4712 said.

trashmaster9000, 2015-09-27:
Ohhh, i have overlooked the Option to Flash Older Versions...

closed 2015-10-01 by sskaug