bitdump/BLHeli #98: ADC Problem for XP7A with 330r resistor

meleeshop, 2015-09-27:
hi everyone,
thank you for working on blheli, actually i find a problem with the xp7a. when i replace the 330r risistor on xp7a with 334r (330r +4r) and the esc will work extra fine as a 330r resistor. usual we would not find the bug, the standard 330r resistor will work with 99% of the motor, but as i start to use a 1410 motor and 2s lipo, the motor started very difficult. with 1s lipo would be OK, problem only for 2s. testing with about 10 motors and 10 escs. even with 1s lipo, the xp7a esc also not good as a xp3a, with xp3a i can start the 1410 motor very soft with very small throttle, but xp7a must be with some vibration. replace the resistor with 334r on f330 chip pin 19, and problem solved. please would someone tell me how to find the code to change the ADC data on pin 19, so i don't need to change the resistor. thanks! PS. with other small size motor you will not find the problem, with a 1306 motor the xp7a will worked very good in 1s or 2s, also 22xx motor would not bring the problem.

sskaug, 2015-09-30:
I am not quite sure I understand what you mean here. Which resistor are you changing? Pin19 is the bemf sensing common node, as far as I can see.
If this is the node, there is nothing the code can do, as a comparator is used to read the node, and not the ADC.

meleeshop, 2015-10-01:
the pin19 connected with 4 resistor, 3x 5.1k resistor, and the 5.1k connected with motor wire A,B,C. there is also a 330r resistor connect to the pin19 and GND. when i change the 330r resistor with 334r, the motor will start much more gently. that's means more volt must be add to the pin19 from the node.

sskaug, 2015-10-01:
This is probably because the other bemf sensing inputs use 1k/5.1k dividers. While the common branch has 330/(3x5.1k) divider. And then, ideally, the 330 ohm should be 333.33 ohm.
Anyway, nothing to do with this in the code, as said it is read by a comparator, not an ADC.
Of course the code can always be better in dealing with such offsets, but that's a challenging task :).

closed 2015-10-06 by sskaug