bitdump/BLHeli #99: New ESC

chestnutbk, 2015-10-04:
Hi Skaug,

I've designed new ESC 20A using C8015F390, with high speed driver mosfet, and modified some lines in code. Can you share me your contact (email) to discuss  with you? My email: chestnut.bkit@gmail.com
[image]

sskaug, 2015-10-04:
Sure! Email sent

closed 2015-10-06 by sskaug

augustinmanecy, 2015-10-20:
Hi Skaug,

I'm a researcher in UAV field, and I'm really interesting to perform closed loop control on rotor speed. Currently, I'm using external hall effect sensor to measure rotational speed of motor, and I adjust the PPM sent to the ESC to perform closed loop control (see http://www.gipsa-lab.fr/projet/RT-MaG/X4-MaG_Documentations.php). This is definitely not an optimal solution, as it needs additional sensors...
I really need this measurement of the rotational speed. Do you think it would be possible to make your BLHeli software transmit this speed via a bus? You can contact me to discuss about this, my email: augustin.manecy@gmail.com

tracernz, 2015-10-20:
BLHeli can already output RPM, see these macros at the end of each board's inc file which just need filled in to set and clear the rpm pin:
```asm
MACRO Set_RPM_Out
ENDM
MACRO Clear_RPM_Out
ENDM
```

Additionally BLHeli already has an RPM governor function, see the governor documentation.

augustinmanecy, 2015-10-21:
Hi,

Ok thank you very much! I will take a look at this RPM_Out. And maybe later
I will add some code to use a serial bus.

Thank you very much,

Augustin