bitdump/BLHeli #100: BLHeli Suite source?

SleepyBrett, 2015-10-23:
I'm fairly new to the scene and I'm not sure if you are the party responsible but perhaps you can put me in touch with them.

I'm a osx developer who is annoyed that I have to keep an xp image around to run blhelisuite. I'd love to take a crack at porting it to osx. Please help me help others like me.

tracernz, 2015-10-23:
See 4712/BLHeliSuite#2

joshuabardwell, 2015-10-24:
If you've got the inclination, port it to Chrome instead, and then everybody can enjoy the goodness.

Trussroads, 2015-10-26:
+1 for the chrome instead of OS ;D

tracernz, 2015-10-26:
Chrome on Windows is a nightmare for USB devices, just see my recent work on DFU flashing in CF configurator or the situation with USB-CDC and CF configurator. If everybody moved on to Linux or OSX then it would be great.

theantnest, 2015-10-26:
Just to let you back down into reality gently... that's never going to happen.

4712, 2015-10-26:
Does it help to make the suite run with Wine?
Is Wine a realistic path for osx?

tracernz, 2015-10-26:
Making it work with Wine would be great. The only thing that seems broken in Wine at first glance is the discovery of COM ports. Unsure if it's realistic for Mac.

tracernz, 2015-10-26:
Looks like Wine is workable on OSX and can be easily installed using Homebrew: http://wiki.winehq.org/MacOSX#head-6d6da43e839bea7e90d199c7eaaa16d92702a7e0

4712, 2015-10-26:
I have no idea how to detect the ports under Wine, so I simply added Com1..99 to the list.
This is what works with "sudo ln -s /dev/ttyUSB0 ~/.wine/dosdevices/com1" so far.
But that is only one problem. The next one is a raised exception because of a read file timeout (Error 1460. There is a workaround possible, but I do not like dirty solutions: I would rather like to know, why the exception is raised...

[image]

4712, 2015-10-28:
BLHeliSuite14.2.0.0 released.
Added support for basic functionality under Linux/Wine (might be also OSX/Wine ). Please see “..\Manuals\BLHeliSuiteWine.pdf” or https://github.com/4712/BLHeliSuite/blob/master/Manuals/BLHeliSuiteWine.pdf

Mac-D3, 2016-02-06:
I added a repository https://github.com/Mac-D3/BLHeliMac for a Mac GUI for BLHeli.
For now, it only supports SiLAB-ESC's and no firmware flashing.Just configuring BLHeli v14.2.

4712, 2016-02-08:
Here's a video on how to use BLHeliSuite on Mac under Wine
https://www.youtube.com/watch?v=OrmrIDIpzJ0

Mac-D3, 2016-02-11:
Thanks for sharing the link to the video. 
I'm sure it works this way, but I (and maybe others like @SleepyBrett) prefer native MacOS apps.

justgook, 2016-02-15:
I think beat solution would be to create Crome app or some kind of it, other solution would be create extension for some kind of FC configurator. Then you be able use same configurator where you set your PIDs jus another tab/section for escs. As I know blhelisuit already support naze32 as bridge for esc, why not try to use same configurator for that CleanFlight or baseflight?

justgook, 2016-02-15:
Btw if u need some help with JavaScript I have some exp (last 7 years working in front end)

AndKe, 2016-04-18:
Hi all, I just got here, just to look for a Linux version (native, not wine)  - just because I am not stuck in the 80's with windows :)   but I agree, Chrome app would most likely be an even better solution..

sl1pkn07, 2016-05-14:
chrome/ium base plugin sounds reasonable for multi-OS purpose (like cleanflight)

i prefer native GUI (Qt if is possible), but this never will happen

wingsuit, 2017-03-23:
How's BLHeli in Wine on Linux looking these days?