bitdump/BLHeli #101: midpoint setting

mkieffer0, 2015-10-29:
Hello everyone.
I tried (for a special application) to set the midpoint to 1300 (in bidir operation)
So far no problem.
But at ppm min throttle input, (1000us) i don't get full reverse power, and full forward power is already set at +- 1800us. (it seems the code is ignoring the min/max throttle settings from the PC interface, and add or subb 500us from the midpoint.
@line 6395 you subtract #125, to check throttle gain.
When i set there different values, eg 200 or 100, i can get my full trottle points either in one or the other motor direction.
To make it work correctly, i must add conditional jumps, if RCP < 1300 or RCP> 1300, and then subb either #200 or #100 from A.
As i'm not so good in assembler, maybe someone has a "cleaner" solution to this.
Any help would be greatly appreciated.

Marc

sskaug, 2015-10-30:
Yes, you are correct. Currently BLHeli uses the same "gain" between throttle signal and motor pwm for both fwd and rev. I guess the best implementation would be to create two gains, one for fwd and one for rev. The code structure should be well suited for doing that.
Another problem is code space. The Afros only have a few bytes left, so probably some functionality needs to be removed. Maybe time has come to remove TX programming...

mkieffer0, 2015-10-30:
We don't use TX programming anyway, and there are a lot of " box " programming possibilities, who are much more convenient and easy to use for the consumer.Can you tell me what range of values are stored in New_RCP @line 3490 (pca_int_limited) using PPM 50 hz input?

joshuabardwell, 2015-10-30:
Please don't remove throttle calibration at the very least. I find this to be a very useful tool. Without throttle calibration, I don't know any other way that I could verify whether an ESC had a bad oscillator, which happened to me recently.

As for all the other options that can be programmed by stick, I agree that stuff can go.

mkieffer0, 2015-10-30:
Back to my first comment,
I tried today @line 6395 to subb A different values (original 125)
subb 200 is working fine to get max throttle at 1000us ppm input.
But values around 100 ( to stretch the gain from 1300 to 2000) provoke a "jump" in motor RPM at nearly full power.
Anyone has an idea why this happens?
Thanks to all.

sskaug, 2016-02-19:
Inactive for a long time, closing for now.

closed 2016-02-19 by sskaug