bitdump/BLHeli #102: Throttle issue

btimby, 2015-10-30:
Hi,

I am not sure this is a BLHeli issue, but it seems like it might be.

I am using Multistar 10A ESCs on a quadcopter.I flashed them to the latest version of BLHeli to resolve sync issues (motors stopped during quick throttle spikes). However, since flashing now the ESCs hold the highest throttle input. For instance after arming then disarming, the motors keep spinning at minthrottle until I remove the battery. If I use the Cleanflight GUI to send a throttle signal to the ESCs, the same thing happens, they ignore lower throttle signals and the motors spin at that throttle until I remove the battery. The motors do no spin until the ESC receives a throttle signal, but then they won't stop spinning.

I am looking for some guidance, thank you.

0n3-70uch, 2015-10-30:
Hello,

have the same problem. I use the 3D function. The motors stop with each change of motor direction. With BLheli 14.1 everything works great.

Greetings
Fabian

0n3-70uch, 2015-10-30:
Here is a short video:
https://youtu.be/taWxmbeao08

btimby, 2015-10-30:
Not sure that is the same problem I am describing.

My ESCs hold the highest throttle position until power is cut.

I will however try some previous versions of BLHeli, as 14.2 is what I flashed before the problem started.

0n3-70uch, 2015-10-31:
Oh yes, it looks like another problem.

btimby, 2015-11-03:
I did figure this out, it was basically user error. When flashing the new firmware (which DID solve my sync issues BTW). Transferring the settings must have fouled up. The input polarity changed from positive to negative. All I had to do was reset this and recalibrate the ESCs.

closed 2015-11-03 by btimby