bitdump/BLHeli #104: Means of recovering from corrupt ESCs

dustin, 2015-11-06:
Three out of four of the ESCs I just hooked up to blhelisuite still work.  The fourth one sings a happy little song and then blhelsuite claims it can't connect.

Is there a way to repair this?  I'd rather not throw away this ESC just because of a software issue.

4712, 2015-11-06:
Did it ever connect? And what type of ESC is it?

dustin, 2015-11-06:
These are littlebees.  Did not ever connect.  However, I was able to flash it using C2.  That was a bit confusing, but I'll take it.  Connecting via C2, everything seemed OK.  I could see the new version of blheli I'd installed, some settings I'd made, etc...  After reflashing it there, I could also reconnect via 1wire.

[image]

readerror67, 2015-11-06:
I have had 2 instances of ESCs getting bricked while flashing. Is the bootloader re-written during normal flashing? Logs show that it writes a bit then fails on a write.

4712, 2015-11-06:
No, bootloader does not override itself, it may happen, that connection is harder to establish or impossible after a flash failed, due to the fact, that the firmware cannot jump to the bootloader. With SiLabs this is more problematic, because the MCU does not have bootloader fuses.

dustin, 2015-11-06:
@readerror67 Are you able to flash them with C2?  It was easier for me than replacing the ESC.

The weird thing for me was how normal everything looked when I connected via C2.  Using the signal wire, it would play music when I tried to connect, but then just fail.  With C2, I saw everything exactly as I'd configured it on the others.  Then reflashed it anyway and then it worked again.

sim-, 2015-11-06:
Hello! Not sure if the same can be done for SiLabs, but I set the reset vector to the end of the program, and jump from there back into the actual code. This way, if the flash is erased and then not fully programmed, it will nop-sled back to the boot loader section. (Won't work as well if replaced page-by-page.)

sskaug, 2015-11-07:
Good input, which could be possible with SiLabs. The flash area 0x1E00-0x1FFF is reserved. The area 0x1C00-0x1DFF is the bootloader. And the area 0x1A00-0x1BFF is settings. So at reset maybe jump to the end off the settings area, and then back?
Would this fit with what the Suite does, Achim?

4712, 2015-11-07:
What Suite does on flash:
1. erase settings page (13)
2. erase and write each page top  to bottom (12 down to 1)
3. write settings page (13)

So, it would be useful to keep reset: section in the first page (afair for 14.1 it is in page 8).
And yes, a jump to the BOOT_START at reset should be fine... as long as the MCU init stuff is done before.

sskaug, 2015-11-07:
Makes sense. But maybe not jump to BOOT_START, because then bootloader will be activated. What about the following:
- Jump to (new) reset vector in the first page
- Jump to BOOT_START minus a few bytes
- Jump to real (current) reset point
Should not be necessary to initialize ports or anything for these two initial jumps.

4712, 2015-11-07:
"Jump to BOOT_START minus a few bytes"
Means there must be some jump code in the settings page?

sskaug, 2015-11-07:
Well, that is one option. But probably we could also jump to just before the settings page (0x1A00 minus a few bytes). This may work since the settings are the last to be programmed.
Just looked at the opcodes, and actually 0xFF isn't nop, it's mov R7, A. But that should works just as well as nop, I guess.

4712, 2015-11-07:
Better jump to just before the settings page... Lets try

4712, 2015-11-15:
Testing with "Jump to BOOT_START minus a few bytes" did not work because of the watch dog reset.
Seems it is enabled by default...
I found the following procedure to work bullet proofed ;)
1. Erase page 13 (pseudo EEprom section to invalidate the ESC against BLHeliSuite 
2. Erase page 1 and write "jmp BOOTLOADER" at page 1 start
3. Erase page 0
4. Erase and write page 2 to 12
5. Write page 0
6. Erase and write page 1
7. Write page 13

I will realize this for the next BLHeliSuite version.

sskaug, 2016-02-19:
Implemented in BLHeliSuite 14.3.0.2

closed 2016-02-19 by sskaug

Alloutofdonts, 2017-01-09:
Is there a way to do this (on a 'bricked' esc) that a simpleton like myself could understand??