bitdump/BLHeli #105: throttle issue blheli 14.2

gaballah, 2015-11-08:
I recently flashed some Xrotor 40A with blheli 14.2. I entered the motors tab in the cf gui afterwards to test my setup. When using the master slide everything worked as expected, but when I used my transmitter to control the throttle 2 out of 4 motors started to increase throttle by themself. they were spinning higher and higher until I unplugged my battery. the throttlestick was at the same level all the time and the remaining two motors were spinning according to the stick input. A friend of mine experienced the same issue with dys 20A ESC´s. We both reflashed our ESC´s with blheli 13.2 but the problem persists. both of our setups were working before we flashed blheli 14.2. He uses frsky equipment and I am using openLRS. 

I wasn´t abled to calibrate the ESC´s through the cf gui either. I  also used my old 32u4 boards with oneshot enabled Multiwii 2.3 firmware. Despite I could calibrate the ESC´s with this board the problem with the motors spinning up by themself was the same.

Any ideas what could be wrong?

best regards

joshuabardwell, 2015-11-08:
This is 100% normal behavior when your copter is on the bench or the ground, and you arm the flight controller. The ground or bench "pushing" on the copter in response to vibrations causes the PID loop to try to push back and correct for the bias. Since the copter is on the ground, the bias cannot be corrected, and the PID loop spins up to an extreme value.

There is nothing abnormal about this situation. Don't arm the copter and then leave it sitting with the motors spinning and the throttle up.

gaballah, 2015-11-08:
Hello,
thank you very much for your reply. 
Your explanation makes total sense to me.  I also just realized it was a pair of motors that would spin up to counteract a rotation around the yaw axis (stupid me)
 I was just wondering because I never experienced this behaviour before and it wasn´t that obvious when using 13.2. and Cleanflight / MW 2.3., or I just didn´t notice..I´m running 14.2 in combination with betaflight now, maybe thats the reason why :-)

Thanks again for clearing that up 
have a nice day.

closed 2015-11-08 by gaballah