bitdump/BLHeli #107: BLHeli has difficulty starting some motors

BertoldVdb, 2015-11-29:
Hello,

I am using EMAX 20A ESCs. They are what I had on hand. I have installed BLHeli on them to drive EMAX MT2204-2300kV motors.

Two out of four motors spin up immediately, but two others take a very long time to start. In around 20% of cases the motor does not start at all. I am wondering why BLHeli gives up after 5 attempts:
  clr C
  mov A, Stall_Cnt
  subb  A, #5
  jc  jmp_wait_for_power_on
  jmp init_no_signal
Is there any risk in removing this test so it will keep trying? 

I also had one ESC+Motor blow up during testing. I think BLHeli is not at fault and the ESC was faulty before. 

Thanks,
Bertold

sskaug, 2015-11-29:
You can surely increase that number from 5. In rev 14.2 it was 10, and before that it was infinite.
As to the starting problems, I've seen several Emax escs with significant mismatch between the three bemf detection branches. Probably cheap components with loose tolerances.

BertoldVdb, 2015-11-29:
Thank you very much. I will be getting a new ESC anyway. Another ESC blew up and this one was not yet flashed with BLHeli so it is clearly a hardware fault.

madwasp, 2015-12-20:
Hello,

I'm experiencing similar issues starting motors also.
I've upgraded some SN20A ESCs with the latest BLHeli firmware and two of the motors (1 and 4) are not starting at the same time of the other two motors.

Also, a friend has exactly the same issue. He has ESCs from other brand, upgraded with the latest firmware version and motors 1 and 4 are starting later than 2 and 3 motors.

Can this behavior be related to some issue in the BLHeli firmware?

Best regards

sskaug, 2016-02-19:
Some improvements to startup implemented in rev 14.4

closed 2016-02-19 by sskaug