bitdump/BLHeli #108: Use github releases

arrowcircle, 2015-12-02:
Hi! Github have releases feature, where you can add any file as release. It supports changelog and versioning. Its also easy to use github api to check for new releases.
Can you consider using it?

KevinVecchione, 2015-12-07:
Would love to see this as well.

rhys100, 2016-06-26:
Please add this! Downloading zip file is slow and does not have good changelog

MBadberg, 2018-10-28:
Wow... nothing here since over two years... The releases feature would be much better, as mediafire.

dentex, 2020-11-16:
I wanted to suggest the same... Why not using GitHub releases? Mediafire or whatever download location is kinda cumbersome.

Thanks

dentex, 2021-08-30:
wow x 2 ... years go by ...

sskaug, 2021-08-31:
Sure, time flies....
Thanks for the reminder!

dentex, 2021-08-31:
> Sure, time flies....
> Thanks for the reminder!

Thanks for taking a look at this!