bitdump/BLHeli #110: ESC XRotor-20,Rev 12.0 is better

leisurers, 2015-12-03:
Compared to the latest firmware(rev 14.0 or above), I found fimeware 12.0 is better suited for esc xrotor-20A, the motor runs more smoothly and is not that "violent" when startup. 
Specially when I used the ESC with pixhwak for my drone,  the motor will not stall when blade is installed.On the contrary, the rev 14.0 may sometimes casue stratup failure.

leisurers, 2015-12-03:
I think the reason is the hardware of Xrotor-20A is not quite match the latest firmware.When I use rev 14.x on Skywalker-40A, the motor will commutate correctly and will not stall. Please excuse my poor English :)

zqbing, 2015-12-04:
你好，你是哪位？xrotor20硬件是我设计的，不知有遇到什么问题？

leisurers, 2015-12-04:
你好，你是好盈的工作人员？是这样的，我分别使用BLHeli 14.0以上版本和12.0版本软件，发现14.0版本软件电机启动扭力比较大，而12.0比较平缓，而且在低油门段下12.0固件电机转速相对低一些。因此在快速从电机停止到拉高油门，或者说飞控解锁时候，当电机带桨时，14.0版本固件在xrotor-20A上面会引起电机无法正确换相，从而无法正常转动。而12.0版本固件就没这个问题。但是这两个固件在Skywalker-40A的硬件上面都不会有这个问题，所以我觉得是不是xrotor-20A的硬件，特别是上面的mos专用驱动芯片不是很好的匹配14.0版本固件。

leisurers, 2015-12-04:
你好，你是好盈的工作人员？是这样的，我分别使用BLHeli 14.0以上版本和12.0版本软件，发现14.0版本软件电机启动扭力比较大，而12.0比较平缓，而且在低油门段下12.0固件电机转速相对低一些。因此在快速从电机停止到拉高油门，或者说飞控解锁时候，当电机带桨时，14.0版本固件在xrotor-20A上面会引起电机无法正确换相，从而无法正常转动。而12.0版本固件就没这个问题。但是这两个固件在Skywalker-40A的硬件上面都不会有这个问题，所以我想14.0版本固件可能不是和xrotor-20A的硬件，特别是上面的mos专用驱动芯片很匹配。

4712, 2015-12-04:
@zqbing and @leisurers: Gentlemen, would you please perform conversations in English language?

leisurers, 2015-12-04:
@4712 Sorry,I will not use home language here any more :-) 
@zqbing my email:leisurers@163.com

4712, 2015-12-04:
...or so a translation as well?

leisurers, 2015-12-04:
I don't quite get it...you want me to translate the words above?

4712, 2015-12-04:
I meant: If you write in Chinese language, add a English translation...

leisurers, 2015-12-04:
This guy @zqbing  designed HW of Xrotor-20A,it's my honor to see you here.

zqbing, 2015-12-06:
Sorry,I designed xrotor 20A hardware,you said you have problems on the esc maybe we can discuss the issue?my english too poor  sorry!

发自我的 iPhone

closed 2015-12-11 by leisurers