bitdump/BLHeli #113: Support for HobbyWing Skywalker 50A

trojanc, 2015-12-15:
Does this have support for Support for HobbyWing Skywalker 50A?

sskaug, 2015-12-15:
No, that ESC has not been mapped.
But most likely it will run with the Skywalker 40A code.

trojanc, 2015-12-15:
I thought of trying it but am too scared im going to brick my ESC. Im too noob to know whats going on.... any one that can help contribute for it? Is there a way I can help without understanding the firmware?

leisurers, 2015-12-18:
Skywalker 50A is same with 40A in mos control part

trojanc, 2015-12-18:
How does these "mappings" work? Is there some way I can connect it and pull some data from the ESC which I can give to one of you guys to do the magic with? Pictures of the components with their labels?

zqbing, 2015-12-19:
yes，please give the PCB NO. Model,  like HW1xx-Vx.x;
i need to confirm the hardware board;

trojanc, 2015-12-21:
Hi

**ESC Description**
Hobbywing Skywalker 50A-UBEC
2-4S Lipo BEC 5v@3A

**PCB details**
HW135_V3.4 (printed on PC)

**Cover details**
HW-BQ8013 (printed on wrapper sticker)

See here for some images: [Google Drive](https://drive.google.com/folderview?id=0BxrHZ_UUpx3MSWRvS09Xa2lEcUU)
(this should allow access for all to view)

I hope this help?
Thanks alot!

closed 2016-02-19 by sskaug

DerElch87, 2016-12-12:
Dear all,
Are there any news about this specific ESC?
The 50A Skywalker is still not listed in the supported Silabs ESC pdf.
The PCB pictures are in the Google Drive folder above my post.
Is there anything I can do to help if it supports BLHeli?
I am only a hobbyist, so I'm pretty unsure about destroying it by try&error...
Thanks in advance.

Tony2k, 2019-02-11:
I'm 2 years late but I still try to improve my Skywalker 50A so I flashed it with SKYWALKER_40A_MAIN_REV14_9.HEX but I wasn't so lucky... Now motor never spin, esc make sound when powered and motor freeze but don't turn. Can I do something?

bastian1982, 2023-06-24:
> Hola a todos, ¿Hay alguna noticia sobre este ESC específico? El 50A Skywalker todavía no aparece en el pdf compatible con Silabs ESC. Las imágenes de PCB están en la carpeta de Google Drive arriba de mi publicación. ¿Hay algo que pueda hacer para ayudar si es compatible con BLHeli? Solo soy un aficionado, así que no estoy muy seguro de destruirlo por prueba y error... Gracias de antemano.

Hola , quiero configurar mi 50A HobbyWing Skywalker como bidireccional. Intento conectarme al BLheli para configurarlo y no tengo caso. Usted me podria indicar los pasos a seguir? lo tengo conectado a una controladora matekf405 con firmware betafligth.- Gracias