bitdump/BLHeli #114: version 14 and above do not work on HKRC 10A (XP-7A)

dkm, 2015-12-16:
My ESC is a 10A from HKRC for my mcpx. I've bit struggling after an update to latest blheli with incorrect behavior. At startup, I get the 3 beeps. When I change throttle, I do get the long beep, but it is immediately followed by the other long beep (ie. it does not wait the 0 throttle). After that sequence, motor won't start. Instead, I get another 3 beeps (same as after poweron) and so on...

After downgrading to 13.2, it works correctly. The erratic behavior happens on all version 14 of blheli.

sskaug, 2015-12-18:
Have you enabled pwm input?
Default rev14 does not recognize pwm input.

dkm, 2015-12-18:
Ok, after enabling pwm input, the arming sequence works. But after that, motor does not spool correctly (hardly makes 2 round then stops). I guess there are some other settings to change, but default values are not working...

sskaug, 2016-02-19:
Inactive for a long time, closing for now.

closed 2016-02-19 by sskaug