bitdump/BLHeli #117: Mux_B definition in DYS_SN20A.inc

nonamerubbish, 2016-01-21:
The section below is in file "DYS_SN20A.inc".

;*********************
; PORT C definitions *
;*********************
.EQU	Mux_B		= 7	; i

the file "m8Adef.inc" is included in file "DYS_SN20A.inc".

but i cannot find any description about PORTC7 or PC7. there are only 0~6 in Port C in file "m8Adef.inc".

sskaug, 2016-02-19:
This is just a placeholder for ADC input 7

closed 2016-02-19 by sskaug

nonamerubbish, 2016-02-26:
However, Mux_B was applied in the MACRO Set_Comp_Phase_B, and this MACRO was applied right under the Label 'set_comp_phase_not1' in the paragraph 'Set comparator phase'. what would happen when this placeholder was called?

nonamerubbish, 2016-02-26:
OK, I find it.

It is not 'Port C definitions'. It is 'ADC definitions'. Because Mux_B is out to the ADMUX register.