bitdump/BLHeli #120: Ques on the code in the beep_onoff routine in the asm for atmel.

nonamerubbish, 2016-02-05:
What is the purpose of the four lines codes right after "beep_onoff"?

It seems to me of no use for the following codes, becauz only FETs of A and B were changed. will the cbr or sbr change the contents of BpFET_off, BnFET_on, AnFET_on, AnFET_off and so on?

well, i did not try to comment those lines and flash the ESC. would anyone pls to try it?

sskaug, 2016-02-19:
Good point. This code was used to change between A and C fets (for stronger beeps), and did that when direction was a part of the AnFET_on etc routines. But direction was removed from the FET_on etc routines, so now the beeping no longer toggles between A and C. I forgot to update this function in the beep when direction was removed from FET_on etc.
I will take a look at fixing in the next rev.

sskaug, 2016-03-31:
Will be corrected in the next rev (probably a rev14.5)

closed 2016-03-31 by sskaug