bitdump/BLHeli #121: Inactivity/lost model beep is capable of frying motors.

aricblunk, 2016-02-19:
Whenever the inactivity beep has been going for more than a couple of minutes, it's possible to catch of whiff of melted laminations from motors. On a previous quad where I'd increased the inactivity beep volume to 160 and took about an hour to find the quad, throttling up the next time caused the motors to puff, showing that the inactivity beep had cooked the laminations and shorted the motor. Rather than having the inactivity beeps be constant, it might be better to only have it beep once every 5 seconds or so, to allow the motors to stay cooler.

sskaug, 2016-02-19:
Thanks for feedback. I'll put looking at it on the list for the next rev

Psi-nz, 2016-02-20:
I also found the motors were getting burning hot.
But i did have the beep strength and beacon strength set to max.
Changed them both to 80 and it seemed to fix the problem.
ESC was SN20A

deepimpactio, 2016-02-20:
I set them to max beep and beacon and the initialization twitch is now extreme and they also heat up noticeably where  they never did before with Rotorgeeks RG20 on version 14.4.0.1

sskaug, 2016-03-31:
I will change the beep rate in the next rev, to about one third the current rate.

closed 2016-03-31 by sskaug