bitdump/BLHeli #124: Betaflight with BLHeli 14.4 Desync?

cylstudio, 2016-02-23:
Please check the issue the group having, thanks.

http://l.facebook.com/l.php?u=http%3A%2F%2Fwww.rcgroups.com%2Fforums%2Fshowthread.php%3Ft%3D2464844%26page%3D1288&h=fAQHKK2LB

4712, 2016-02-23:
Sorry, your link does not work.

cylstudio, 2016-02-23:
repost
http://www.rcgroups.com/forums/showthread.php?t=2464844&page=1288

sskaug, 2016-02-23:
That thread is a bit hard to collect data from. Can you summarize what the problem is, what esc, motor, prop, battery etc?

PiotrKlimczak, 2016-02-23:
Hi,
Just noticed this thread.
I am also facing desync issues on 14.4 with:
4s lihv 15.2v, Afro 30a opto, emax 3510 600kv, 13x5.5 props, naza m v2. 
With default settings awful desyncs. 
Switched off demag and damped light all fine. 
Haven't tried reenabling any of them as I am scared of crash as desyncs were happening only with props and start time. 
3.05kg hex.

sskaug, 2016-02-24:
I'm not sure what this is, and I don't have your motor. There are many motors out there...
But I do have a Multistar W48-22 KV690, and I do have a 13x4.7 prop. Running it at mid charged 4S battery, it pulls 35A at full throttle. Using Afro 30A with default settings rev14.4 (where damped light is default), it runs just fine.

closed 2016-05-22 by sskaug