bitdump/BLHeli #125: 14.4 desyncs with damped light and demag on naza m v2, Afro 30a, emax mt3510, lihv 4s

PiotrKlimczak, 2016-02-23:
1st of all really great work on blheli. 
I've moved from dji 30a opto to Afro just for blheli- both were widely recommended. 

But as in subject with blheli 14.4 started facing desyncs (some engines suddenly started to spin slower and strange way) with damped light and demag on naza m v2, Afro 30a, emax mt3510, lihv 4s, hexa 3.05kg with battery. 
It doesn't happen without props. Only when engines are spinning up with props close to take off.
It is perfectly stable without demag and damped light.
Don't have testing desk unfortunately.
I am scared of testing enabling damped light, as I have almost crashed on 1str try with default settings. 
Is the testing desk only option?
Is this known problem?

sskaug, 2016-02-24:
I'm not sure what this is, and I don't have your motor. There are many motors out there...
But I do have a Multistar W48-22 KV690, and I do have a 13x4.7 prop. Running it at mid charged 4S battery, it pulls 35A at full throttle. Using Afro 30A with default settings rev14.4 (where damped light is default), it runs just fine.

PiotrKlimczak, 2016-02-24:
Thanks for letting me know. 
At least I know my Afro escs should work on 4s. Suspecting it shouldn't be naza fault too.
So only motors stayed to blame.

PiotrKlimczak, 2016-02-26:
Just ordered servo tester and have some plans how to test engines with props not necessarily flying. 
Will share findings once will find something

PiotrKlimczak, 2016-03-19:
OK, so servo tester proven stability with Damped Light on and Demag off.
Tested on one "arm" of my hexacopter. So decided to take a flight.
When spinning up to take off again- it failed, then again, and again.... no success.

New finding is that actually it is happening only to 2 out of 6 engines, where for one of them very often.
Also what i have noticed, occurrences of the problem dropped down significantly after "heating" motors/escs a little- simply by spinning them for some longer time on close to take of rpm. After that I even had few successful flights without single desync.

Now in such case, what can be the cause?
Any idea?
Thanks in advance.

sskaug, 2016-03-19:
Why not demag?
Setting it to low or high will reduce the likelihood of sync loss at low rpms.

PiotrKlimczak, 2016-03-20:
Power wires are as short as they can be in 700mm frame.
Anyway I crashed today ugly and decided never try this option anymore. But 1st has to rebuild my hexacopter.
Damed light is definitely not worth trying for non 3D and even very dangerous as in my case.
I suggest switching this option off as default in firmware as it puts people and their aircraft at risk.

closed 2016-03-20 by PiotrKlimczak