bitdump/BLHeli #127: Brake at zero throttle (glider)

tcalvert, 2016-02-29:
Hello,

I have been using BLHeli on all my multirotors for some time now and love it :+1: 

Recently I built a glider and used a spare FVT20A ESC from my 250. This works great, except that when I chop the throttle, the motor doesn't stop.

I'd like to be able to have the motor brake when I cut the throttle, so that my prop will fold. Right now it keeps on spinning with the airflow, which causes a lot of drag :(

I dug into the source code, but I'm not totally sure were to change stuff in BLHeli.asm, any pointers would be appreciated. I also tried various configs (closed loop mode etc.), but it seems like the FETs are just all switched off when idle.

Thanks for your work on this great software!

sskaug, 2016-03-01:
Great to hear that it works well for you.

"call switch_power_off" is a key routine. It turns off all fets.

"run_to_wait_for_power_on_stall_done:" label is where the code exits from running. Here power is also switched off. So if you, after power is switched off, turn on some fets (eg all nfets) then the motor will brake. As far as I can see, switch_power_off is called again before restarting the motor.

tcalvert, 2016-03-01:
Thanks for the pointers Steffen, just what I was after.

I forked the repo and made the change you suggested, although I'm not sure it will work with all the PWM signals disabled (does PWM go to the high or low FETs?)... This evening I will try to get it to compile, and I'll see how it goes.

Once I have a working version I will see about adding a 4th value to the "PWM frequency" config (after damped light) to enable this behaviour, and will consider submitting a PR.

tcalvert, 2016-03-08:
Finally I got a version working. In the end the changes I ended up keeping were these : https://github.com/tcalvert/BLHeli/commit/e9cccb660f367819a8394f80eb0bb75c591e09b2.

Also, since I used Keil's Simplicity Studio, I couldn't get the main file to compile (possibly due to its size?). I got it working by removing all the hardware conditional code in the first 1000-2000 lines. This version as well as the HEX file can be found here : https://github.com/tcalvert/BLHeli/tree/master/Vendor%20specific%20versions/TCALVERT

Many Thanks Steffen for not only making this great firmware, but also for taking the time to maintain it and respond to requests like mine :+1:

closed 2016-03-08 by tcalvert

htirc, 2016-03-10:
@sskaug Hello, Steffen! Many thanks for your amazing works! Recently, we are developing our own BLHeli ESC. But we got some questions which we can't load Htirc_20A.HEX/EEP to BLHeliSuite14.4.01. Could you please tell us what we should do to solve it? If possible, we'd like to send you more details we have now. Sincerely hope to get your kind reply! Thank you very...very much for your valuable time!

SlipstreamFPV, 2016-03-30:
I'd really like this feature in the standard BLHeli setup as well. A lot of the new escs, the small escs and the most efficient escs are now made to run BLHeli and a lot of them even come with BLHeli on them from the factory so the only option for using them on gliders and some other planes would be implementing this feature :) I hope you will reconsider adding it sskaug! In particular I've been using the xrotor 40A quite a lot.

tcalvert, 2016-03-31:
@SlipstreamFPV indeed most of the ESCs are made to run BLHeli now. I think that it would be nice to have something like MODE=3 (PLANE), which would include such changes. Because after flying my version a few times, I've noticed that the braking is not sufficient when I chop the throttle at high speeds, it would be required to keep on doing the regenerative braking (damped light) until the motor is spinning slowly, and only then brake using All_nFETs_On.

For the moment I guess you could compile my changes for the XRotor 40A and have something that works for a glider...

ilihack, 2016-05-14:
Hello, the Brake at zero Trottle can help my Copter if the pixhawk fail to slow down the free fall down a little bit or not? it is a very interesting Feature thx :)

tcalvert, 2016-05-16:
@ilihack No, in fact this will have the opposite effect. It is only useful for airplanes and gliders.

ilihack, 2016-05-16:
okay thank you for you replay. but why opposite Effekt? it fall faster from
the air ?