bitdump/BLHeli #130: Suggestion: Integration of cleanflight passthrough for SimonK bootloader

rav-rav, 2016-03-18:
After putting together code from [https://github.com/cleanflight/cleanflight/commit/fa862f4395a76e1bd05f425a19edd461df4c9cc0](cleanflight usblinker) and [https://github.com/c---/ArduinoUSBLinker/blob/master/ArduinoUSBLinker.ino](ArduinoUSBLinker) I have written a passthrough feature for betaflight.

Right now it is difficult to use, but it could be integrated in BLHeli Suite just like the 1wire passthrough. Any thoughts?
https://github.com/rav-rav/betaflight/commit/f948ea37ef18aaf71a0b29dff7fedb1245f4da8e

tracernz, 2016-03-18:
Was done some time ago https://github.com/sambas/cleanflight/blob/master/src/main/drivers/serial_escserial.c#L824

4712, 2016-03-18:
@tracernz: With help of your integration of vcp baud-rate Beside of some a lot cleaning up needed and a random hang after disconnecting for unknown reason, I got vcp pass through working fine. 
For the 256k flash devices a 4way-interface port (excluding C2) is available, but it costs some kb.
I will post it soon.

tracernz, 2016-03-18:
@4712 Nice! I look forward to trying it.

rav-rav, 2016-03-18:
Great solution, I didn't know it existed. Looking forward to seeing this integrated into BLHeli Suite.