bitdump/BLHeli #132: 4-in-1 ESCs

davidbitton, 2016-03-22:
Could the codebase be adapted to a [C8051F363](http://tro.pe/1WE13BB) chip? It runs at 100 MHz and has 5 digital ports. All of the 4-in-1 ESCs that I see are really just 4 ESCs "glued" together. Would be nice to reduce the chip count and have one MCU for all four motors. Is this feasible? Thanks!

sskaug, 2016-03-22:
Still, at a minimum, I'd say four comparators would be needed. 
Not sure about other recources, like memory etc.

davidbitton, 2016-03-22:
I checked and SiLabs does NOT have an MCU with 4 comparators; Atmel does. The three supporting processors are [here](http://tro.pe/1Zpa6rD). The only difference between the three Atmel chips is memory size (as I can see).

davidbitton, 2016-04-01:
@nonamerubbish can you be a little more specific so I know what to look for in an MCU? This is from the C8051F363 [datasheet](http://tro.pe/1RR4Z3H).

> 1.7. 10-Bit Analog to Digital Converter
The C8051F360/1/2/6/7/8/9 devices include an on-chip 10-bit SAR ADC with up to 21 channels for the differential
input multiplexer. With a maximum throughput of 200 ksps, the ADC offers true 10-bit linearity with
an INL and DNL of ±1 LSB. The ADC system includes a configurable analog multiplexer that selects both
positive and negative ADC inputs. Ports1-3 are available as an ADC inputs; additionally, the on-chip Temperature
Sensor output and the power supply voltage (VDD) are available as ADC inputs. User firmware
may shut down the ADC to save power.

Is this what you mean?

nonamerubbish, 2016-04-02:
In fact, i just start learning the AVR MCU. i don't know much about C8051. so the method i provide  could be possible, but not must be possible.

the zero cross event is checked with **Analog Comparator**, but not **Analog and Digit Convertor**, in the BLHeli program for AVR MCU as i know. is it the same for C8051 MCU?(@ sskaug)